micropython/micropython #2238: esp8266 socket.getaddrinfo("0.0.0.0", port)

todork, 2016-07-13:
socket.getaddrinfo("0.0.0.0", port) restart device for micropyton version > v1.8.1

==================================================================================
MicroPython v1.8.2-11-gbe313ea-dirty on 2016-07-13; ESP module with ESP8266
Type "help()" for more information.
>>> import socket
>>> addr_info = socket.getaddrinfo("0.0.0.0", 8266)
Fatal exception 3(LoadStoreErrorCause):
epc1=0x4020ff25, epc2=0x00000000, epc3=0x00000000, excvaddr=0x402773c6, depc=0x00000000

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 30612, room 16 
tail 4
chksum 0x94
load 0x3ffe8000, len 1044, room 4 
tail 0
chksum 0x49
load 0x3ffe8420, len 2540, room 8 
tail 4
chksum 0xc2
csum 0xc2
��
  ��O|�	��sOs`
                      ���dG��c@sGx�c@sGN��d��
                                               ��l�s�l l��sOy{Os`
                                                                    �dl���c@rG��c@sGN�	ox

                                                                                           ��
                                                                                             l`	rF����O|�	��sOs`
                                                                                                                      ���
                                                                                                                         G��c@cG��c@rGN�	gwG��c@sGd�`	cO�8��	�
 ������No�dd0������No�dd0���d�s��	�7F��N�c���cs�cvc�N�Io�Io�cNlcp�gΓN�FF�	o�cpy
                                                                                                      ��
                                                                                                        �d��
==================================================================================

dpgeorge, 2016-07-13:
Works for me.  Please update to the latest esp-open-sdk and rebuild.

todork, 2016-07-13:
I'll check

ст. н.с. II ст.,
ст. чл. IEEE,
д-р Тодор Куюмджиев
Изп. Директор ЕКоКоМС ООД
тел.:    +359 2 9743929
факс.    +359 2 9743986
Уеб:     www.ecocoms.com
Скайп:   tkouyoumdjiev
twitter: @tkouyoumdjiev
          @ECoCoMS

Senior Research Fellow,
Senior Member IEEE,
Dr. Todor Kouyoumdjiev
CEO ECoCoMS Ltd.
Tel.:    +359 2 9743929
Fax.     +359 2 9743986
Web:     www.ecocoms.com
Skype:   tkouyoumdjiev
twitter: @tkouyoumdjiev
          @ECoCoMS

На 13.07.2016 в 12:02, Damien George написа:

todork, 2016-07-13:
Thanks a lot Damien!

That was the point.

I started with vanilla install of esp8266-micropython-vagrant and things 
worked.

Thanks again for the assistance.

Regards,
Todor

ст. н.с. II ст.,
ст. чл. IEEE,
д-р Тодор Куюмджиев
Изп. Директор ЕКоКоМС ООД
тел.:    +359 2 9743929
факс.    +359 2 9743986
Уеб:     www.ecocoms.com
Скайп:   tkouyoumdjiev
twitter: @tkouyoumdjiev
          @ECoCoMS

Senior Research Fellow,
Senior Member IEEE,
Dr. Todor Kouyoumdjiev
CEO ECoCoMS Ltd.
Tel.:    +359 2 9743929
Fax.     +359 2 9743986
Web:     www.ecocoms.com
Skype:   tkouyoumdjiev
twitter: @tkouyoumdjiev
          @ECoCoMS

На 13.07.2016 в 12:02, Damien George написа:

closed 2016-07-13 by dpgeorge