micropython/micropython #2239: reserved identifier violation

elfring, 2016-07-13:
I would like to point out that identifiers like "[`__MICROPY_INCLUDED_LIB_UTILS_PYHELP_H__`](https://github.com/micropython/micropython/blob/master/lib/utils/pyhelp.h#L26)" and "[`__INET_NTOP_H`](https://github.com/micropython/micropython/blob/b8945517723efaafa63df9ce52bb9d3fa3fceda5/drivers/cc3000/inc/inet_ntop.h#L1)" [do eventually not fit](https://www.securecoding.cert.org/confluence/display/c/DCL37-C.+Do+not+declare+or+define+a+reserved+identifier#DCL37-C.Donotdeclareordefineareservedidentifier-NoncompliantCodeExample%28HeaderGuard%29 "Do not use identifiers that are reserved for the compiler implementation.") to the expected naming convention of the C language standard.
Would you like to adjust your selection for unique names?

dpgeorge, 2017-07-18:
@elfring thanks for bringing this issue to our attention.  It's now been fixed, see 299bc625864b9e624ed599c94a5f95870516139a

closed 2017-07-18 by dpgeorge

elfring, 2017-07-18:
Thanks for [your improvement of include guards](https://github.com/micropython/micropython/commit/299bc625864b9e624ed599c94a5f95870516139a "Unify header guard usage.").

elfring, 2019-02-09:
I have accidentally found [another update candidate](https://github.com/micropython/micropython/blob/b9ec6037edf5e6ff6f8f400d70f7351d1b0af67d/py/misc.h#L50 "Update MP_STRINGIFY?").

dpgeorge, 2019-05-09:
Thanks @elfring, should be addressed by c0a1de3c216e86e679520fa50daf41dd9ac3ea78

elfring, 2019-05-09:
Thanks for [another small source code adjustment](https://github.com/micropython/micropython/commit/c0a1de3c216e86e679520fa50daf41dd9ac3ea78 "Rename _MP_STRINGIFY").