micropython/micropython #2240: esp8266 better PWM

deshipu, 2016-07-13:
I will just leave those links here for the future. There is a nice PWM implementation for the ESP8266. Would it make sense to switch to it at some point?

https://lurchi.wordpress.com/2016/06/29/esp8266-pwm-revisited-and-reimplemented/
https://github.com/StefanBruens/ESP8266_new_pwm

mcauser, 2016-07-22:
Related thread: http://www.esp8266.com/viewtopic.php?f=4&t=10600

hemanth7787, 2017-12-26:
https://hackaday.io/project/12933-esp8266-geiger-counter/log/44406-micropython-10khz-soft-pwm

robert-hh, 2017-12-26:
Works. Very useful. I hope I can recall that when I need it.

dpgeorge, 2020-06-09:
The esp8266 could do with improved PWM, but the driver linked at the top is GPL, so a no-go.

The 10kHz patch linked above is interesting (and simple) but changes the maximum duty from 1023 to 100 so reduces the resolution.

I'll close this issue for now.

closed 2020-06-09 by dpgeorge