micropython/micropython #2243: ADC not working for STM32L4

mattbrejza, 2016-07-14:
The ADC peripheral does not work on the L4, calling pyb.ADC(channel).read() always returns 0.

I've tried to fix it by comparing ST examples, and I've managed to get a bit more life out of it with the following change in [adc_read_channel](https://github.com/micropython/micropython/blob/master/stmhal/adc.c#L191):
```
if (HAL_ADC_PollForConversion(adcHandle, 10) == HAL_OK){
   if ((HAL_ADC_GetState(adcHandle) & HAL_ADC_STATE_REG_EOC) == HAL_ADC_STATE_REG_EOC)
		rawValue = HAL_ADC_GetValue(adcHandle);
}
```

and `sConfig.SingleDiff = ADC_SINGLE_ENDED;` in the channel config.

Now, if you read a channel 1-16 it returns ~310, (with about +/- 10 variance on this), whatever channel is selected. The exception to this are the internal channels  CH17,18, which return ~900 and ~800 respectively.

This suggests that the channel selection isnt working correctly, however I am unable to find the issue.

tobbad, 2016-07-19:
Hi
I created a pull request #2252 for fixing this issue.

It would be great if you could test this fix by checking out and compiling `l4_adc_bug` branch of my repo
https://github.com/tobbad/micropython/tree/l4_adc_bug
If the fix works on your device too, please add some kind of "OK" to Issue #2252.
Tobias

mattbrejza, 2016-07-19:
Hi Tobias,

Thanks for looking at this. I have tested it, it appears to work fine, except the L4 has its 16 GPIO channels labelled 1-16 rather than 0-15. As a result, the ADC_IN16 GPIO input cannot be read.

The first thing that will need changing is L127 in adc.c, and also the code which generates pins_BOARD_NAME.c, since pin_adc1[] does not contain the pin for ADC16

mattbrejza, 2016-07-19:
Another thing I have found, is that [this check](https://github.com/tobbad/micropython/blob/5b9f7c987b262945c9b6afabee9a1328ed3d563c/stmhal/adc.c#L256) should only be done if channel <16 (f4) or only if 0<channel<=16 (l4).

As it stands, for the L4 you cant do pyb.ADC(0).read() to read the voltage reference

dhylands, 2016-07-19:
I think that the pins in pins_BOARD_NAME.c should match the data sheet. Looking at the definitions in the L4 hal files, it looks like ADC_CHANNEL_1 maps to a value of 1 for both L4 and F4.

So we can either increase the 16 to 17 in make-pins.py for all boards (which would consume 3 more words of flash on the F series, or add something to mpconfigboard.mk which Makefile reads and then passes an additional argument to make-pins.py

mattbrejza, 2016-07-20:
The alternative is to put a `#if defined(MCU_SERIES_L4)` around the 17th element

dhylands, 2016-07-20:
> The alternative is to put a #if defined(MCU_SERIES_L4) around the 17th element

Yeah - that sounds like a reasonable alternative as well.

tobbad, 2016-07-21:
I propose, that we leave the PR as it is because it fixes the bug which it references. And @mattbrejza opens a feature improvement PR for using all 16 ADC channels with the solution resulting from the discussion here.

From the feedback I got from the PR I did, I have the feeling, that you should not mix Bugfix and feature improvements PRs.

dpgeorge, 2016-08-11:
The fix for the non-working ADC was merged.  Is anyone still interested in getting the 16th channel working?

mattbrejza, 2016-08-11:
I have the fix working, I just need to put it in a PR now the original PR has been merged

dpgeorge, 2016-08-25:
ADC16 should now be available on the L4 MCUs.

closed 2016-08-25 by dpgeorge