micropython/micropython #2249: Micropython won't run main.py when reset in binary REPL

matoha, 2016-07-18:
Hello,
to reproduce this issue,

1. Enter binary mode (I used Ctrl+A)
2. Soft-reset MP - either Ctrl+D or sys.exit(), doesn't matter

now Micropython doesn't know any of the functions in main.py and I have to run 
`from main import *` to get them working again.

When reset in normal REPL, everything works as intended, this affects only binary REPL.

Version: 1.8.2, STM32F405

dpgeorge, 2016-07-18:
This is by design.  Raw REPL is intended to be used by programs (eg running the test suite), not by humans, and in that case you don't want main.py to run each time you do a soft reset.  If you need to execute main.py then make sure you are in friendly REPL before issuing a soft reset: (ctrl-B, ctrl-D, ctrl-A) will work.

matoha, 2016-07-18:
OK, thanks. Actually, I have it connected to a program (data acquisition) and at the beginning of a new cycle I do a reset.

I will modify my main.c to remove the REPL condition in if.

`if ((reset_mode == 1 || reset_mode == 3) && pyexec_mode_kind == PYEXEC_MODE_FRIENDLY_REPL) {`

closed 2016-07-18 by matoha