micropython/micropython #2255: Could not open file 'boot.py' could not open file 'main.py' for reading

mflmartin, 2016-07-23:
I just successfully flashed the latest version esp8266-2016-07-23-v1.8.2-39-g2dd21d9 to an ESP-01 (Black) with 1MB of Flash memory, with: 

```
(esp8266) D:\Sync\Martin\inProgress>python D:\Sync\Martin\inProgress\_virtualenvs\esp8266\Scripts\esptool.py -p COM6 -b 460800 write_flash --flash_size=8m 0 D:\Sync\Martin\inProgress\esp8266\esp8266-2016-07-23-v1.8.2-39-g2dd21d9.bin
esptool.py v1.1
Connecting...
Running Cesanta flasher stub...
Flash params set to 0x0020
Writing 512000 @ 0x0... 512000 (100 %)
Wrote 512000 bytes at 0x0 in 11.6 seconds (352.4 kbit/s)...
Leaving...
```

But it seems the filesystem never got expanded, therefore it never reads boot.py and main.py.

Serial output:

```
▒#4 ets_task(40100368, 3, 3fff62f0, 4)
could not open file 'boot.py' for reading
could not open file 'main.py' for reading

MicroPython v1.8.2-39-g2dd21d9 on 2016-07-23; ESP module with ESP8266
Type "help()" for more information.
>>>

```

However, I do have 1MB and it seems propertly configured:

Output from diag_port:

```
 >>> import port_diag
FlashROM:
Flash ID: 1440e0 (Vendor: e0 Device: 4014)
Flash bootloader data:
Byte @2: 00
Byte @3: 20 (Flash size: 1MB Flash freq: 40MHZ)

Networking:
STA ifconfig: ('0.0.0.0', '0.0.0.0', '0.0.0.0', '208.67.222.222')
AP ifconfig: ('192.168.4.1', '255.255.255.0', '192.168.4.1', '208.67.222.222')
Free WiFi driver buffers of type:
0: 8
1: 0
2: 8
3: 4
4: 7
lwIP PCBs:
Active PCB states:
Listen PCB states:
TIME-WAIT PCB states: 
```

pfalcon, 2016-07-23:
What issue is being reported here?

mflmartin, 2016-07-23:
Failure to expand the filesystem or set itself correctly.

Or am I missing something? If the esp has more than 1Mb flash, it should
flash correctly with a file system.

pfalcon, 2016-07-23:
> Failure to expand the filesystem or set itself correctly.

So, where's failure? Info provided by you doesn't show anything even remotely resembling a failure.

> Or am I missing something?

You apparently write the above with some unspoken assumptions, but these assumptions aren't clear to other people, e.g. me. Try to make them explicit.

mflmartin, 2016-07-23:
Hello pfalcon, 

I am sorry if I was not clear enough, or specific enough. I followed all the installation process in the esp-01, as stated in the documentation in a esp8266-01. 

**When I try to use any function in that setup that involves the filesystem, I just get errors.** But the board has 1MB memory and is propertly flashed. 

Like for instance
```
>>> os.mkdir("hello")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 13] EACCES

```

or this ( wich outputs that, and then seems to reboots the serial connection and the esp8266 (as you see in the serial output specified here, it tries again to read the main.py and boot.py):
'''
>>> os.listdir()
['\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\                                                              x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x0                                                              0\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\                                                              x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00                                                              \x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x                                                              00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00                                                              \x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x0                                                              0\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\                                                              x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00                                                              .\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x                                                              00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00                                                              ', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x0                                                              0\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\                                                              x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x0                                                              0\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x                                                              00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00                                                              \x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x                                                              00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\                                                              x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00',                                                               '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x                                                              00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00                                                              \x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x                                                              00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\                                                              x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x0                                                              0\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\                                                              x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00                                                              \x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x                                                              00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.                                                              \x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x0                                                              0\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00'                                                              , '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00                                                              \x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x                                                              00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00                                                              \x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x0                                      [... cut ...] x00\x00.\x00\x00\x00']
>>> os.listdir()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
**MemoryError: memory allocation failed, allocating 1024 bytes**
'''

I apologize if I was not specific enough. Because of the above tests I think the filesystem doesn't expand correctly although it has the room for it and the board was flashed with the corresponding flag.

Or you still think I am missing something completely? (which can be the case, I don't say the contrary)

Thanks,

mflmartin, 2016-07-23:
I just flashed esp8266-2016-07-23-v1.8.2-40-g0d22177.bin, and it works. 

**esp8266-2016-07-23-v1.8.2-39-g2dd21d9** didn't work. 

So chances are there **was indeed an induced bug in that version**, as I said, based on empiric testing. If I flash that one, it doesn't work. If I flash the newly released esp8266-2016-07-23-v1.8.2-40-g0d22177.bin, it works. ( Or, it is a miracle. )

closed 2016-07-23 by mflmartin

pfalcon, 2016-07-23:
> Like for instance

~~~
>>> os.mkdir("hello")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 13] EACCES
~~~

Well, your original description didn't include that info, and that of course clarifies it.

> I just flashed esp8266-2016-07-23-v1.8.2-40-g0d22177.bin, and it works. esp8266-2016-07-23-v1.8.2-39-g2dd21d9 didn't work. ( Or, it is a miracle. )

It likely is - there were no changes which could explain that. Please confirm that you got those binaries from http://micropython.org/download/#esp8266 . Also, if you can, please attach esp8266-2016-07-23-v1.8.2-39-g2dd21d9.bin here - as it's gone from download page, I can't test it myself.

Otherwise, here's what's currently known about such issues: http://forum.micropython.org/viewtopic.php?f=16&t=1908 (last question of the FAQ as of now). Please keep us posted if you face similar issue with a newer daily build again (and keep esp8266-2016-07-23-v1.8.2-40-g0d22177.bin handy, to use it as a known-working reference).

mflmartin, 2016-07-23:
Hello Again, 

I see, you are right. I should have included that example in the first place for clarity's sake.

I just found other users with the same problem in the forum:

https://forum.micropython.org/viewtopic.php?p=12285#p12285

I provided an alternative and possible solution, because from time to time my esptool fails when flashing and keeps stopping at 99%, so it might be related (but I am unsure, because I can't reproduce it now with the new version).

Here is the bins that you asked for, if you can try them out in a ESP8266-01 with 1MB flash:

[esp8266-2016-07-23-v1.8.2-39-g2dd21d9.zip](https://github.com/micropython/micropython/files/379957/esp8266-2016-07-23-v1.8.2-39-g2dd21d9.zip)