micropython/micropython #2258: esp8266: Max reliable I2C freq?

mcauser, 2016-07-23:
I managed to get my DHT12 I2C sensor working on my WeMos D1 mini by lowering the freq argument when constructing the I2C object from 400,000 to 50,000. Anything higher was flakey.

I was using the default 400,000 as per the quick ref docs.
http://docs.micropython.org/en/latest/esp8266/library/machine.I2C.html

I was able to use baud=400,000 when running a similar script on my pyboard v1.0.

Seems to be speed related - software I2C vs hardware I2C.
Is there a max reliable SCL frequency for the software I2C implementation on the ESP8266?

Troubleshooting here:
http://forum.micropython.org/viewtopic.php?f=16&t=2168&p=12279#p12279

deshipu, 2016-07-23:
If you specify too high a frequency for the bit-banged i2c, it will simply just go as far as it can without reaching that frequency you specified. In other words, unless the device in question has a limit on the frequency, or your connection is inadequate in some way (like too weak pullup resistors), there is no way you can break anything with too high i2c frequency.

pfalcon, 2016-08-09:
Extending explanation above, max reliable frequency depends on i2c device, cable length and pull resistor values (and quality), ambient temperature, and of course, phase of the moon. Find one which works for your case, or use standard 100KHz if in doubt. Feel free to share your findings on the forum.

closed 2016-08-09 by pfalcon