micropython/micropython #2259: Integer math exceptions in interrupt handlers

noelhenson, 2016-07-23:
This issue may be a bug. It may also just be something that users need to be aware of when writing interrupt handlers.

I have been having issues on several projects where performing simple math in interrupt handlers causes "MemoryError: memory allocation failed, heap is locked" exceptions. In my cases, I have been using pyb.micros() timer values to make relative timing measurements between events. The problem is appeared to be intermittent but is actually based on the timer values when pyb.micros() is near maxint (0x3fffffff).

The problem shows with both variable and array integers:
x = 0
or
x = array.array('i',[0])

Repro:

import intException
intException.doit()

This should yield this error message:

```
Uncaught exception in ExtInt interrupt handler line 4
Traceback (most recent call last):
  File "intException.py", line 17, in testint
MemoryError: memory allocation failed, heap is locked

```

[intException.zip](https://github.com/micropython/micropython/files/379956/intException.zip)

pfalcon, 2016-07-23:
pyb.micros() is deprecated, see http://docs.micropython.org/en/latest/pyboard/library/utime.html#utime.ticks_us for replacement designed specifically to solve this problem. But mind https://github.com/micropython/micropython/issues/2217

closed 2016-07-23 by pfalcon

noelhenson, 2016-07-23:
I should have been more clear. I encountered the issue when I was doing math with values from pyb.micros(). The actually issue involves heap exceptions during simple math operations in interrupt handlers.

```
# Integer math exception repro

import pyb
import array
import micropython

micropython.alloc_emergency_exception_buf(100) # interrupt debugging

y = array.array('i',[0])

x = 0x3fffffff
y[0] = 0x3fffffff

def testint(line):
    global x,y
    # uncomment only one of the two following lines
    x = x+1
    #y[0] = y[0] + 1

MYINT = pyb.ExtInt(pyb.Pin.cpu.C4, pyb.ExtInt.IRQ_RISING_FALLING, pyb.Pin.PULL_UP, testint)

def doit():
    MYINT.swint()
```

dhylands, 2016-07-23:
The problem is that only small-int integer math is supported in interrupt handlers. small-ints are typically represented as a 31-bit signed entity. Any number which can't be stored in a small-int needs to be converted to big-int, which requires memory allocation.

0x3fffffff + 1 = 0x40000000 which can't be represented in a 31-bit signed number.

pfalcon, 2016-07-23:
Sure, the documentation should specify that you can't perform allocations in irq handlers, and working with long ints require allocation. So, you should make sure to stay in small int range.

noelhenson, 2016-07-23:
I understand.

It was my mistaken expectation that maxint + 1 would have wrapped to 
-maxint while keeping its small-int type; especially when using 
array('i') storage.

Thanks,

Noel

peterhinch, 2016-07-24:
@noelhenson Python isn't a strongly typed language and integers have arbitrary precision. So if you add two integers you will get the mathematically correct value regardless of the number of bits required. If you need a result constrained to a particular number of bits, it's your job to achieve this. In the case of MicroPython there are occasions where you need to be mindful of its internal representation of small integers.

noelhenson, 2016-07-24:
Peter,

Thank you for taking the time to investigate and explain. I had hoped 
that the using an array.array("i") would have worked since it is a fixed 
type. Being an ancient assembly language programmer, I was used to 
values that wrap.

I am glad its not a bug but a knowledge issue instead.

Noel