micropython/micropython #2260: NUCLEO_F767ZI port issues

jeremyherbert, 2016-07-25:
Hi,

As a way of getting a bit more acquainted with the micropython codebase, I have ported (it compiles, but not yet tested, so there could be more changes needed) micropython to the STM32F767ZIT6 (2048K flash, 512K SRAM) on the new NUCLEO board. I have come across a few issues when doing this, and I am looking for some guidance as to how I can overcome them so I can submit my code back to the master branch.

The main problem is that this chip requires a new version of stm32hal, so I needed to replace those files. It seems as though the stm32f7xx_hal_sd.h/.c files were modified to rename some of the functions inside, meaning that a straight copy of files is not enough for an upgrade. Is there an important reason why these renames were not done using inline function aliases and/or defines outside of the HAL library files? 

The new stm32hal also requires a new APB/AHB prescaler table to be added the system_xxx.c file for the STM32F7 series (I haven't checked if this is the case for the F4). Is there a particular reason why both the F7 and F4 share a manually merged version of this file? It seems like they will probably diverge in the future, and this may hinder upgrading to new versions of stm32hal.

dhylands, 2016-07-25:
The stm32 hal functions have a few bugs. We document all of the hal changes made to the HALs in this file:
https://github.com/micropython/micropython/blob/master/stmhal/hal/HALCOMMITS

This commit, https://github.com/micropython/micropython/commit/1d7fb82f0aec11331635532583617d773888b991 was the one affecting the SD files.

We renamed the functions to make it obvious that we're using modified versions and to make a new hal fail (since the new hal probably doesn't properly support 64-bit).

The system_xxx (we use this file: https://github.com/micropython/micropython/blob/master/stmhal/system_stm32.c) is just a sample file, it isn't part of the HAL.

Whether to use a new file or merge additional stuff is something that is done on a case by case basis.

dhylands, 2016-07-25:
The PR which discusses the SDHC changes is here: https://github.com/micropython/micropython/pull/849

dpgeorge, 2016-08-11:
@jeremyherbert if you're still interested in adding support for the F767 (note the F746 is already supported) then the first thing would be to upgrade the HAL in the stmhal/hal/f7 directory.  But before that is done I would like to understand why the F7 HAL needs upgrading?

jeremyherbert, 2016-08-11:
@dpgeorge I am definitely still interested in doing it, in fact I think it is done. But I'm in the middle of writing up a thesis so all fun projects are currently on the backburner ;)

My recollection was that a bunch of defines were missing, probably due to the fact that the version in micropython was last updated before the chip was released. I will try to check up on this later today. I did try to compile just by replacing everything in the F746 code with F767 specific, but it wasn't that simple.

Maybe it would be a good idea to git submodule out the stm32 hal (in a new repo under the micropython organisation), so it would be easier to see changes?

dpgeorge, 2016-08-11:
> My recollection was that a bunch of defines were missing, probably due to the fact that the version in micropython was last updated before the chip was released.

Yes, I see that there is no F767 in the CMSIS files which are more critical than the HAL, so they'd need updating as well, plus the addition of the cmsis/devinc/stm32f767xx.h file (I assume that's what it's called).

> Maybe it would be a good idea to git submodule out the stm32 hal (in a new repo under the micropython organisation), so it would be easier to see changes?

We discussed that already when adding STM32L4 support and decided it was simpler to just add the files to this repo.

dhylands, 2016-08-11:
I'm not sure how putting the hal in a submodule makes it any easier to see the changes?

We document all of the changes we make to the HAL in this file:
https://github.com/micropython/micropython/blob/master/stmhal/hal/HALCOMMITS

and in git you can just cd into the hal directory and do:
```
git log .
```
to get a change history of just the files in that directory tree.

jeremyherbert, 2016-08-11:
@dhylands I just thought it would make sense to have issues, etc separated out. If someone was interested in the Teensy port for instance, seeing the stm32hal commits in the list is just noise really. For someone who isn't familiar with the codebase but would like to contribute, I haven't seen something being organised like this before. Are you supposed to commit to the HALCOMMITS file in a separate commit after the code commit? I don't think there is a way to know the commit hash beforehand.

It doesn't really bother me either way, just thought I would make the suggestion.

@dpgeorge Yes, I updated the CMSIS stuff too.

dhylands, 2016-08-11:
If you do ```git log .``` in the stmhal/hal directory you won't see any teensy commits or any other commits for files outside that directory.

We normally check in the virgin HAL files, and then check in the changes, and then update HALCOMMITS

jeremyherbert, 2016-08-11:
Ok thanks. I will do that once I get back around to working on this.