micropython/micropython #2261: Bogus UTF-8 handling

abiliojr, 2016-07-25:
Running the following pieces of code in MicroPython would produce quite different results to what it would be expected (CPython 3):

```python
a = b'\x80abc'
print(a.find(b'a')) # prints 1, as expected
print(a.find(b'a', 1)) # should print 1, but instead prints -1
```

Now, the next code should print 1, instead it prints 2:
```python
print(b'\x80aa'.find(b'a', 1))
```

I think these two previous examples could be grouped in a single bug.

I had gone into modifying the function str_index_to_ptr found in the objstrunicode.c file, adding the condition `&& type != &mp_type_bytes` to the two lines that call UTF8_IS_CONT into the str_index_to_ptr function.

That worked to fix the bug, but then I realised the problem could be more general, and that's how I found this other issue:
```python
a = '¢' # this is an UTF-8 char
print(len(a)) # prints 1, as expected
print(a[0]) # prints ¢, as expected
print(a[1] == '\x00') # prints True, in real world it should throw: index out of range
```

Being new around MicroPython internals, I prefer to report my findings here, so more expert minds could figure out the cleanest solution to fix them.

The original problem was found in a WiPy, but they will also happen on the Mac OS port of MicroPython, so I think is a general behaviour.

pfalcon, 2016-07-25:
Thanks for the report. 1st issue is fixed in 6af90b29725a85f275380d1973b1454e25e6bdbc. Subscription issue is to be investigated.

pfalcon, 2016-07-25:
The subscription issue is also fixed in ed1c194ebf12c1e894d7620ca89c70f87b5d5709.

closed 2016-07-25 by pfalcon