micropython/micropython #2263: esp8266 micropython-lib

biggosh, 2016-07-25:
Hi,

I'd like to know how to install some modules from micropython-lin (http) on esp8266.
I've never compiled micropython to use on this device.

thanks in advance
BG

peterhinch, 2016-07-26:
You may need to install git on your PC. Use git clone URL to download micropython-lib to your own machine. Then put the library files you need onto the ESP. You don't need to compile MicroPython as the lib files are Python code.

It's possible to use wget to download individual files, but it's useful to have the entire library in local storage and git makes it easy to retrieve updates for the entire library with git pull.

biggosh, 2016-07-26:
thank you very much, so simple I didn't though to.

closed 2016-07-30 by pfalcon