micropython/micropython #2264: FROZEN MPY problem

dmazzella, 2016-07-26:
I have used hmac from micropython-lib and on unix this work fine, but when i have tried to use FROZEN MODULES the result is different.

copied python scripts from /c/Users/Damiano/OneDrive/micropython/micropython-lib to /home/dma/.micropython/lib/

without FROZEN MODULES and with /home/dma/.micropython/lib:

```
$ ls -l /home/dma/.micropython/lib/
totale 61
-rw-r--r-- 1 dma None     1 26 lug 11.46 __init__.py
-rw-r--r-- 1 dma None   162 26 lug 11.46 binascii.py
drwxr-xr-x 1 dma None     0 26 lug 11.46 collections
-rw-r--r-- 1 dma None   763 26 lug 11.46 functools.py
drwxr-xr-x 1 dma None     0 26 lug 11.46 hashlib
-rw-r--r-- 1 dma None 10019 26 lug 11.46 hmac.py
-rw-r--r-- 1 dma None   100 26 lug 11.46 io.py
-rw-r--r-- 1 dma None  1829 26 lug 11.46 os.py
-rw-r--r-- 1 dma None 20228 26 lug 11.46 pyb.py
-rw-r--r-- 1 dma None   120 26 lug 11.46 re.py
-rw-r--r-- 1 dma None  1564 26 lug 11.46 string.py
-rw-r--r-- 1 dma None    98 26 lug 11.46 struct.py
-rw-r--r-- 1 dma None  3487 26 lug 11.46 types.py
-rw-r--r-- 1 dma None  3282 26 lug 11.46 uint.py
```

```
$ /home/dma/projects/micropython/unix/micropython.exe
MicroPython v1.8.2 on 2016-07-26; linux version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>>
paste mode; Ctrl-C to cancel, Ctrl-D to finish
=== import hmac
=== import hashlib
=== msg = b'zlutoucky kun upel dabelske ody'
=== dig = hmac.new(b'1234567890', msg=msg, digestmod=hashlib.sha256).hexdigest()
=== print('c735e751e36b08fb01e25794bdb15e7289b82aecdb652c8f4f72f307b39dad39')
=== print(dig)
===
c735e751e36b08fb01e25794bdb15e7289b82aecdb652c8f4f72f307b39dad39
b'c735e751e36b08fb01e25794bdb15e7289b82aecdb652c8f4f72f307b39dad39'
>>>

```

with FROZEN MODULES and without /home/dma/.micropython/lib:

```
$ ls -l /home/dma/.micropython/lib/
ls: impossibile accedere a '/home/dma/.micropython/lib/': No such file or directory

make -j12 FROZEN_MPY_DIR="/c/Users/Damiano/OneDrive/micropython/micropython-lib"
...
MPY /c/Users/Damiano/OneDrive/micropython/micropython-lib/binascii.py
MPY /c/Users/Damiano/OneDrive/micropython/micropython-lib/collections/defaultdict.py
MPY /c/Users/Damiano/OneDrive/micropython/micropython-lib/collections/deque.py
MPY /c/Users/Damiano/OneDrive/micropython/micropython-lib/collections/__init__.py
MPY /c/Users/Damiano/OneDrive/micropython/micropython-lib/functools.py
MPY /c/Users/Damiano/OneDrive/micropython/micropython-lib/hashlib/sha224.py
MPY /c/Users/Damiano/OneDrive/micropython/micropython-lib/hashlib/sha256.py
MPY /c/Users/Damiano/OneDrive/micropython/micropython-lib/hashlib/sha384.py
MPY /c/Users/Damiano/OneDrive/micropython/micropython-lib/hashlib/sha512.py
MPY /c/Users/Damiano/OneDrive/micropython/micropython-lib/hashlib/__init__.py
MPY /c/Users/Damiano/OneDrive/micropython/micropython-lib/hmac.py
MPY /c/Users/Damiano/OneDrive/micropython/micropython-lib/io.py
MPY /c/Users/Damiano/OneDrive/micropython/micropython-lib/os.py
MPY /c/Users/Damiano/OneDrive/micropython/micropython-lib/pyb.py
MPY /c/Users/Damiano/OneDrive/micropython/micropython-lib/re.py
MPY /c/Users/Damiano/OneDrive/micropython/micropython-lib/string.py
MPY /c/Users/Damiano/OneDrive/micropython/micropython-lib/struct.py
MPY /c/Users/Damiano/OneDrive/micropython/micropython-lib/types.py
MPY /c/Users/Damiano/OneDrive/micropython/micropython-lib/uint.py
MPY /c/Users/Damiano/OneDrive/micropython/micropython-lib/__init__.py
```

```
$ /home/dma/projects/micropython/unix/micropython.exe
MicroPython v1.8.2 on 2016-07-26; linux version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>>
paste mode; Ctrl-C to cancel, Ctrl-D to finish
=== import hmac
=== import hashlib
=== msg = b'zlutoucky kun upel dabelske ody'
=== dig = hmac.new(b'1234567890', msg=msg, digestmod=hashlib.sha256).hexdigest()
=== print('c735e751e36b08fb01e25794bdb15e7289b82aecdb652c8f4f72f307b39dad39')
=== print(dig)
===
c735e751e36b08fb01e25794bdb15e7289b82aecdb652c8f4f72f307b39dad39
b'bee2b62eb0c5ca5956c78572f0515c4042bc51f0a7e9b359a82dd4e7646a790d'
>>>

```

peterhinch, 2016-07-26:
It would help anyone trying to fix this if you could identify the module causing the problem when frozen. Trial and error/binary chop could surely isolate it.

dmazzella, 2016-07-26:
The problem seems to be hmac, i have tried only hashlib (frozend and not) and is ok.

pfalcon, 2016-07-26:
See https://github.com/micropython/micropython-lib/pull/82 which may be related or not. The gentleman who submitted it needs some motivation to split up that mixed-up patch in a series of short nice clean patches.

dmazzella, 2016-07-26:
I tried @bynds version but the result is the same.

dmazzella, 2016-07-27:
I tried the hashlib tests and these also fail if FROZEN.

frozen version:

```
MicroPython v1.8.2 on 2016-07-27; linux version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> from hashlib.sha256 import test as sha256_test
>>> from hashlib.sha512 import test as sha512_test
>>> sha256_test()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "hashlib/sha256.py", line 255, in test
AssertionError:
>>> sha512_test()
>>>
```

non frozen version:

```
$ ./micropython.exe
MicroPython v1.8.2 on 2016-07-27; linux version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> from hashlib.sha256 import test as sha256_test
>>> sha256_test()
>>> from hashlib.sha512 import test as sha512_test
>>> sha512_test()
>>>
```

dpgeorge, 2016-08-15:
Should be fixed by 095e43a9a5b3fc7c1220cdbabe6a4fed0973b1e1.

Issue was that bignums (big integers) were being used as indices in a slice object, and that case wasn't handled internally by uPy.  It occured for frozen bytecode (but not normal bytecode) because the frozen code was making bignums where actually one could fit a smallint.

closed 2016-08-15 by dpgeorge