micropython/micropython #2265: esp8266 micropython-lib umqtt.simple API simplification?

peterhinch, 2016-07-26:
The `MQTTClient` class callback appears unnecessary. The `wait_msg` method, if a message is available, runs the callback immediately prior to returning `None`. This is the case whether or not the socket (and hence the method) is set to blocking. The callback is never called asynchronously.

It would seem simpler to eliminate the callback and return (topic, msg). The caller would need to check the type of the return value. The change entails no evident loss in performance or functionality.

pfalcon, 2016-07-26:
Did you read documentation (README), commit messages, original ticket with requirements/discussion?

peterhinch, 2016-07-26:
The API design justifies the callback on the basis that the callback may occur asynchronously, which is the classic justification for using one. However, reading and running the code, it seems that asynchronous callbacks never actually occur. The callback is only executed immediately prior to a return from the `wait_msg` method.

pfalcon, 2016-07-31:
"Asynchronous" is such an overused, broad-meaning word, that possibly its facet which you mean currently is what confuses you.

> The callback is only executed immediately prior to a return from the wait_msg method.

Yes, and wait_msg() isn't always called by user code, to receive a subscribed message, it's called by internal parts of the module as a simple grep shows.

peterhinch, 2016-07-31:
OK, I found your use of a callback unfamiliar. My encounters have been in asynchronous hardware driven events such as timer expiry or UI interactions. Apologies for not adequately following the code.

closed 2016-07-31 by peterhinch