micropython/micropython #2268: esp8266 micropython-lib umqtt.simple subscribe method throws exception

peterhinch, 2016-07-27:
On occasion the `subscribe` method throws an exception because the `sock.read()` call sometimes returns `None`. With method modified as below, when this occurs the `subscribe()` call is successful. So far I've failed to determine the circumstances in which this occurs or to create a simple test case but this code  regularly produces output:
```python
        self.sock.write(qos.to_bytes(1))
        resp = self.sock.read(5)
        if resp is None:
            print('subscribe: None received')
            return
```
The method calls use the default qos. Server is a raspberry Pi running mosquitto as a daemon. mosquitto_pub and mosquitto_sub produce expected results.

peterhinch, 2016-07-29:
Perhaps this offers some clues. I added/uncommented print statements to the subscribe method:
```python
    def subscribe(self, topic, qos=0):
        print(topic, qos, self.sock)
        pkt = bytearray(b"\x82\0\0\0")
        self.pid += 1
        struct.pack_into("!BH", pkt, 1, 2 + 2 + len(topic) + 1, self.pid)
        print(hex(len(pkt)), hexlify(pkt, ":"))
        self.sock.write(pkt)
        self._send_str(topic)
        self.sock.write(qos.to_bytes(1))
        resp = self.sock.read(5)
        if resp is None:
            print('subscribe: None received')
            return
        print(resp)
        assert resp[0] == 0x90
        assert resp[2] == pkt[2] and resp[3] == pkt[3]
        if resp[4] == 0x80:
            raise MQTTException(resp[4])
```
With the following outcome from two subscriptions to 'pyboard' and 'green':
```
b'pyboard' 0 <socket state=2 timeout=0 incoming=0 off=0>
0x4 b'82:0c:00:01'
subscribe: None received
b'green' 0 <socket state=2 timeout=0 incoming=0 off=0>
0x4 b'82:0a:00:02'
subscribe: None received
```
As stated before the subscriptions are successful. I'm happy to add further diagnostics if it's any help.

pfalcon, 2016-07-30:
micropython-lib is a separate project with its pwn bug tracker. And this bug tracker is already very overcrowded (such overcrowded that it's hard to find "real" bug to work on, or at least select one really important). Nothing needs to be done now, the whole point is that *all future* submissions should be done where they belong.

pfalcon, 2016-07-30:
Otherwise, this issue is rather weird. None from read() is returned when it's non-blocking socket and there's nothing top read. None should never be returned for blocking socket. So, there's either bug in umqtt.simple which sometime leaves socket in non-blocking state, or bug in modlwip's read(). You can help with diagnosing both of these possibilities - either review umqtt.simple code carefully, or produce an independent testcase for 2nd possibility.

peterhinch, 2016-07-31:
OK, I think I see where the problem is - in umqtt.simple. I'll do some testing and report back (in the correct place).

pfalcon, 2016-07-31:
Let's continue there then.

closed 2016-07-31 by pfalcon