micropython/micropython #2269: uctypes does not support values larger than 31 bits despite listing uint64 support

ryannathans, 2016-07-28:
```
>>> struct = uctypes.struct(uctypes.addressof(buffer), {"bug": uctypes.UINT32, "bug64": uctypes.UINT64})
>>> struct.bug = 2147483647
>>> struct.bug = 2147483648
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OverflowError: overflow converting long int to machine word
>>> struct.bug64 = 2147483647
>>> struct.bug64 = 2147483648
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OverflowError: overflow converting long int to machine word
```

Seems pretty funny how ctypes can't support UINT32 and UINT64s?

pfalcon, 2016-08-25:
To fix this, we need getters for getting Python int object as: int32, uint32, int64, uint64, each apparently "truncated" and "range checking" version. That's many getters, so apparently there should be one new function, taking flags for the above, deprecating previously available mp_obj_int_get_truncated() & mp_obj_int_get_checked().

dpgeorge, 2016-08-26:
Just using mp_obj_int_get_truncated should fix the 32 bit case.  Then there are already some functions to handle wider values, eg int_to_bytes() and mp_obj_int_to_bytes_impl().

pfalcon, 2017-04-20:
Ok, I have fixes for these working on 32-bit little-endian, now the most boring part is writing tests...

pfalcon, 2017-04-21:
Fixed by 9e8f3163924c1f429f16f44a4a27b0cd33064719

closed 2017-04-21 by pfalcon