micropython/micropython #2272: uctypes hangs system when using float32 not aligned to 4 bytes

ryannathans, 2016-07-28:
```
>>>import uctypes
>>>buffer = bytearray(100)
>>> f = uctypes.struct(uctypes.addressof(buffer), {"bug":uctypes.UINT8 | 0, 'bug2':uctypes.UINT8 |1, 'bug3':uctypes.UINT8 |2, 'bug4':uctypes.UINT8 |  3, 'bug5':uctypes.FLOAT32 | 4})
>>> f.bug = 1
>>> f.bug2 = 2
>>> f.bug5 = 1.1
>>> f = uctypes.struct(uctypes.addressof(buffer), {"bug":uctypes.UINT8 | 0, 'bug2':uctypes.UINT8 |1, 'bug3':uctypes.UINT8 |2, 'bug4':uctypes.FLOAT32 | 3})
>>> f.bug = 1
>>> f.bug4 = 1.1
```
System does not recover and hangs indefinitely 

surprisingly enough UINT32s don't have the same issue

dhylands, 2016-07-28:
Preliminary research suggests that accessing unaligned floats on a Cortex M4 causes Hard Faults. Lots of processors have quirks like that. On X86 you can access using any alignment. On 32-bit ARM under linux, they have a trap handler you can enable which allows the access to continue (you take quite a performance hit though).

I don't see this as a fault of uctypes.

If you really need your code to be portable, you should copy unaligned floats into an aligned float using a bytearray and then access as a float (and this is what you need to do in C too).

ryannathans, 2016-07-28:
@dhylands Perhaps uctypes should be extended to support unaligned floats.

If the float is not aligned, it could be copied to a register, operated on and copied back to the unaligned struct. Thoughts?

pfalcon, 2016-08-25:
Technically, it's not uctypes hangs your system, but your instructing it to do unaligned float access. If your systems hangs on it (others don't), you apparently shouldn't do that.

> Perhaps uctypes should be extended to support unaligned floats.

uctypes support accessing packed (== with alignment of 1, aka unaligned) structures from the beginning: http://docs.micropython.org/en/latest/pyboard/library/uctypes.html#uctypes.LITTLE_ENDIAN . If you have issues with layout types uctypes.LITTLE_ENDIAN or uctypes.BIG_ENDIAN, please provide a testcase and all relevant information to reproduce an issue.

closed 2016-08-25 by pfalcon