micropython/micropython #2276: error: unknown type name ‘ssize_t’

kopchik, 2016-07-30:
Hi!

I just cloned micropython and had the following problem on my arch linux (AMD64, gcc (GCC) 6.1.1 20160707):
~~~
$ make -C mpy-cross
[skip]
CC ../py/modcollections.c
CC ../py/modgc.c
CC ../py/modio.c
In file included from ../py/modio.c:32:0:
../py/stream.h:88:1: error: unknown type name ‘ssize_t’
 ssize_t mp_stream_posix_write(mp_obj_t stream, const void *buf, size_t len);
 ^~~~~~~
../py/stream.h:89:1: error: unknown type name ‘ssize_t’
 ssize_t mp_stream_posix_read(mp_obj_t stream, void *buf, size_t len);
 ^~~~~~~
../py/stream.h:90:1: error: unknown type name ‘off_t’
 off_t mp_stream_posix_lseek(mp_obj_t stream, off_t offset, int whence);
 ^~~~~
../py/stream.h:90:46: error: unknown type name ‘off_t’
 off_t mp_stream_posix_lseek(mp_obj_t stream, off_t offset, int whence);
                                              ^~~~~
make: *** [../py/mkrules.mk:47: build/py/modio.o] Error 1
make: Leaving directory '/home/exe/github/micropython/mpy-cross'
~~~

The fix is trivial:

~~~
--- a/py/stream.h
+++ b/py/stream.h
@@ -26,6 +26,7 @@
 #ifndef __MICROPY_INCLUDED_PY_STREAM_H__
 #define __MICROPY_INCLUDED_PY_STREAM_H__
 
+#include <unistd.h>
 #include "py/obj.h"
 #include "py/mperrno.h"

~~~

mcauser, 2016-07-30:
Getting the same error compiling HEAD on ESP8266

pfalcon, 2016-07-30:
This issue should be fixed now for all ports.

closed 2016-07-30 by pfalcon