micropython/micropython #2278: More reliable esp8266 binary downloads

pfalcon, 2016-07-30:
Some points were brought up elsewhere:

1. If a build fails, there's no binary available for download (should keep previous around).
2. In general, it would be nice to keep 3-4 last builds, to let users and maintainers to catch/compare behavior in case of recently introduced/non-deterministic bug.

deshipu, 2016-07-30:
If you are editing that page, it would also be great to add a link to flashing instructions right there.

dpgeorge, 2016-08-04:
I have implemented the following changes to the download page:

- last 4 daily builds are kept for esp8266 port
- removed hyphens from date format in all firmware filenames to make filenames easier to read
- added shortcut links at the top of the page to the firmware sections
- added link in the esp8266 port section to flashing instructions in the tutorial

In a few days we will see if the changes are all working.

pfalcon, 2016-08-09:
This was implemented (may need tweaks/adjustments to handle esp8266 releases).

closed 2016-08-09 by pfalcon