micropython/micropython #2279: STM32L476 - DAC not working

bobricius, 2016-07-30:
Hi I have try compile micropython for STM32L476 for my nucleo board but I get this error

dac.c: In function 'pyb_dac_deinit':
dac.c:255:36: error: 'DAC_CR_BOFF1' undeclared (first use in this function)
         DAC_Handle.Instance->CR |= DAC_CR_BOFF1;
                                    ^
dac.c:255:36: note: each undeclared identifier is reported only once for each function it appears in
dac.c:258:36: error: 'DAC_CR_BOFF2' undeclared (first use in this function)
         DAC_Handle.Instance->CR |= DAC_CR_BOFF2;
                                    ^
../py/mkrules.mk:47: recipe for target 'build-LIMIFROG/dac.o' failed
make: *** [build-LIMIFROG/dac.o] Error 1

dhylands, 2016-07-30:
I'm assuming that you enabled the DAC? Because the default board configuration seems to build ok.

STM often changes things from chip to chip. So somebody needs to take a look at the datasheets for the F4 and see what DAC_CR_BOFF1 and DAC_CR_BOFF2 do on the F4 series and figure out what the equivalent functionality is for the L4.

dpgeorge, 2018-12-04:
This looks like it was fixed in dcfd2de5c2319f026e53c0a63495823a5bb11706.  The STM32L476DISC board now compiles with DAC enabled.

closed 2018-12-04 by dpgeorge