micropython/micropython #2280: unix memory leak

peterhinch, 2016-07-31:
Also on Pyboard, ESP8266. Pasted at the REPL:
```python
import pickle, gc, micropython
gc.collect()
micropython.mem_info()
count = 0
def bar():
    global count
    s = pickle.dumps(('result', str(count)))
    count += 1
    return pickle.loads(s)
for _ in range(1000):
    a = bar()
gc.collect()
micropython.mem_info()
```
This produces the following outcome with used increasing:
```
mem: total=10303, current=3127, peak=4908
stack: 4992 out of 80000
GC: total: 2072832, used: 4064, free: 2068768
 No. of 1-blocks: 70, 2-blocks: 10, max blk sz: 9
12997
mem: total=2260180, current=376780, peak=378293
stack: 4992 out of 80000
GC: total: 2072832, used: 21248, free: 2051584
 No. of 1-blocks: 74, 2-blocks: 10, max blk sz: 161
>>> 
```

peterhinch, 2016-08-01:
I've narrowed this down a little further. Replacing pickle with json in the above code removes the leak. The following code leaks: I suspect the exec statement.
```python
import gc, micropython
gc.collect()
micropython.mem_info()
count = 0
def bar():
    global count
    count += 1
    s = repr(('result', str(count))) # pickle.dumps
    d = {} # pickle.loads.
    exec("v=" + s, d) #  Leak seems to be here
for _ in range(1000):
    bar()
gc.collect()
micropython.mem_info()
```
It only leaks if s changes on each iteration - the counter is necessary.

dpgeorge, 2016-08-04:
Confirmed.  The reason for the increased memory usage is string interning: the string you are evaluating is something like `v=('result', '123')`, with the 123 changing on each iteration.  The `'123'` is a small string that the parser interns when it parses the code you give to eval/exec/compile.  So as iterations go on there the interned strings from previous runs remain and eventually you run out of RAM.

A work-around for the above script is to replace `str(count)` with `count`, since then it's not a string but an integer.  But that doesn't help the general case.

deshipu, 2016-08-04:
I wonder if string interning could be limited to just string literals that appear in the programs? Would that make sense?

dpgeorge, 2016-08-04:
> I wonder if string interning could be limited to just string literals that appear in the programs?

As far as the parser is concerned, the input from a file is equivalent to the input from `exec`.  So there would need to be extra logic to distinguish these.

deshipu, 2016-08-04:
Ah, you are of course, right, in this case this is actually a string literal being evaluated, I missed that, sorry.

peterhinch, 2016-08-04:
Thanks for that. I'll figure out if I can apply the workround in my application.

dpgeorge, 2022-06-23:
Closing due to inactivity.  Also #4422 tracks a similar issue.

closed 2022-06-23 by dpgeorge