micropython/micropython #2287: Surprising behavior for separator argument to ubinascii.hexlify()

pyhys, 2016-08-02:
When setting the _sep_ argument to an empty string for the ubinascii.hexlify() function, there is still a separator inserted. It is the NULL string "\x00", which has the length 1.

If this is the intended behavior, then the documentation should be updated. A less suprising behavior would be not to insert any separator when the _sep_ argument is an empty string (the same as when not using the _sep_ argument).

This is how it behaves now:

    >>> result = ubinascii.hexlify(b"abc")
    >>> result, len(result)
    (b'616263', 6)
    >>> 
    >>> result = ubinascii.hexlify(b"abc", " ")
    >>> result, len(result)
    (b'61 62 63', 8)
    >>> 
    >>> result = ubinascii.hexlify(b"abc", "")
    >>> result, len(result)
    (b'61\x0062\x0063', 8)

I read issues #993, #1272, #1277 without understanding what the intended result is when the separator is set to an empty string.

pfalcon, 2016-08-02:
The indented behavior is that people *won't* use this incompatible extension. However, if you, at your own risk, use it, you either use a separator string of length 1, or enjoy the risk mentioned (passing a string of anything but 1 char ASCII is undefined behavior).

closed 2016-08-02 by pfalcon

gpshead, 2019-05-20:
Python isn't supposed to have undefined behavior. Either treat an empty string as no sep (expected behavior) or raise an exception if you want to disallow that.