micropython/micropython #2288: Site down?

mcauser, 2016-08-03:
I am unable to reach the MicroPython site or forum.
https://micropython.org
https://forum.micropython.org

http://www.downforeveryoneorjustme.com/micropython.org
http://www.downforeveryoneorjustme.com/forum.micropython.org

Responding to pings though.
```
Mikes-iMac:~ mike$ ping micropython.org
PING micropython.org (176.58.119.26): 56 data bytes
64 bytes from 176.58.119.26: icmp_seq=0 ttl=55 time=308.465 ms
64 bytes from 176.58.119.26: icmp_seq=1 ttl=55 time=285.738 ms
64 bytes from 176.58.119.26: icmp_seq=2 ttl=55 time=280.610 ms
```

mnelsoneorm, 2016-08-03:
Same here.....and for some reason my local webrepl is problematic today too :(

mcauser, 2016-08-03:
@dpgeorge http://dpgeorge.net/ is also down

dpgeorge, 2016-08-03:
Thanks for the report.  I'm on it.

dpgeorge, 2016-08-03:
Ok, everything should be back to normal.

dothacker495, 2016-08-03:
Oh Great, Big thanks. It's like the micropython docs is life itself. It is hard to find other references

pfalcon, 2016-08-03:
@dothacker495 : You can download PDF docs to use locally. And you can build HTML docs yourself too.

closed 2016-08-03 by pfalcon