micropython/micropython #2289: Compilation problems for unix port on OSX 10.11

eduvik, 2016-08-03:
I'm trying to compile unix/ on OSX 10.11. I cam `make axtls` successfully, but the subsequent `make` causes the following errors:

`GEN build/genhdr/qstr.i.last
In file included from ../lib/berkeley-db-1.xx/btree/bt_close.c:48:
In file included from ../lib/berkeley-db-1.xx/PORT/include/db.h:46:
../lib/berkeley-db-1.xx/PORT/include/compat.h:132:9: error: 'memmove' macro redefined [-Werror,-Wmacro-redefined]
#define memmove(a, b, n)        bcopy(b, a, n)
        ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.11.sdk/usr/include/secure/_string.h:70:9: note:
      previous definition is here
#define memmove(dest, src, len)                                 \
        ^
1 error generated.`

lapsule, 2016-08-11:
Same proplem on v10.11.6

dpgeorge, 2016-08-11:
@pfalcon maybe we disable btree for Mac builds?

pfalcon, 2016-08-24:
Trivial fix is to add `#undef memmove` there. Deeper issue is that BDB uses memmove everywhere, including where it's not needed. I already started replacing it with memcpy(), and that would rather continue.

blmorris, 2016-08-24:
For context, here is the section of `/usr/include/secure/_string.h` which contains the conflicting macro definition on my system (OSX 10.11.6)

```C
#if __has_builtin(__builtin___memmove_chk) || defined(__GNUC__)
#undef memmove
#define memmove(dest, src, len)                                 \
  __builtin___memmove_chk (dest, src, len, __darwin_obsz0 (dest))
#endif
```

pfalcon, 2016-08-25:
Pushed fix to master, please test.

blmorris, 2016-08-25:
Works for me- unix port builds, tests pass. Thanks!

pfalcon, 2016-08-25:
Thanks for confirming.

closed 2016-08-25 by pfalcon