micropython/micropython #2291: ESP8266 tests vfs_fat_ramdisk.py

peterhinch, 2016-08-04:
Perhaps I'm jumping the gun with a feature not yet implemented or am otherwise missing something, but this fails here on the reference board. Firmware built from source dated 28th July.

It formatted the filesystem on the ESP8266 flash (beware). It successfully wrote and read back the test file - but it was located in Flash, not on the ramdisk. It then issued an assertion failure (presumably because it hadn't written to the ramdisk).

I reduced the size of the ramdisk in case there were memory issues, but to no effect.

dpgeorge, 2016-08-04:
It's not yet implemented.  ESP8266 can't have more than one filesystem mounted at a time.  Multiple mounted filesystems requires #2140 or similar.

peterhinch, 2016-08-04:
OK, thanks.

closed 2016-08-04 by peterhinch