micropython/micropython #2293: esp8266: Make esp.osdebug(None) the default

tdicola, 2016-08-05:
I noticed with the current esp8266 builds there's a lot of debug output.  There are previous issues about this like #2072 and a mention that perhaps debug output should be suppressed for later firmware builds (see https://github.com/micropython/micropython/issues/2072#issuecomment-219035113 ).  I'd like to suggest that this behavior to suppress debug output become the default now (perhaps just add a call to esp.osdebug(None) to the _boot.py?).

The reason I raise this issue is that scripts like pyboard.py depend on talking to a 'clean' REPL with no debug output.  The esp8266 codebase right now can print out a lot of debug output, like if the chip is trying to connect to a wifi network it can't find, and this wreaks havoc with pyboard.py's parsing which depends on getting exact response strings back from the chip.  If debug output sneaks in it's game over and pyboard.py's parsing fails.

I'd like to suggest debug output suppression be the current new default for esp8266 firmware.  This will allow folks to use pyboard.py and build tooling, etc. around it to improve the ecosystem.  I realize folks can just call esp.osdebug(None) themselves but IMHO it would make the experience much smoother if people can load the firmware and start using tools immediately, not troubleshoot peculiar issues because their esp is in an odd state spewing debug output.  Thanks!

pfalcon, 2016-08-06:
Thanks for the suggestion, but that's exactly how it is, as explained in the FAQ: http://forum.micropython.org/viewtopic.php?f=16&t=1908 . Specifically: release versions have debugging disabled, while development versions (daily master) has it enabled, as required (well, grossly helpful) for development.

> This will allow folks to use pyboard.py and build tooling, etc. around it to improve the ecosystem.

Building tooling is very welcome and appreciated. And one productive way to do it is against the latest release, as otherwise MicroPython ESP8266 is under heavy development and big, and even incompatible, changes may come up any time. And people who do development need logging on by default, so if there's a random segfault, they know it's a segfault and where it happened, not a random lockup (without debug logging they look the same). When the port becomes more stable and developed, default in master can be revisited (but the root of bugs and instability is the vendor SDK, not MicroPython).

In the meantime, I made this change: https://github.com/micropython/micropython/commit/6de37864a27481578a699cb0eaaaccd46bb494ea to give an extra hint to people who want to disable it in development builds.

tdicola, 2016-08-09:
Oh perfect, that sounds great and apologies I didn't realize the release builds disable it.  Cool! :)

closed 2016-08-09 by tdicola

pfalcon, 2016-08-09:
@tdicola : A publicly available release version is now available at http://micropython.org/download/#esp8266