micropython/micropython #2294: esp8266: New Flash map

pfalcon, 2016-08-05:
Vendor SDK 2.0.0 requires another flash sector to store params, natural to extend 4-sector system area at the end of flash to 5 sectors. Taking need to move FS due to this, would be nice to plan ahead future needs for flash. We need extra flash:

1. To accommodate firmware growth.
2. To provide executable area for @native code storage.
3. To leave few sectors for user usage.

Thus proposed: to start filesystem at 0x90000, up from 0x89000 used currently. That will allow to 0xe000 executable chunk, 32K of which could be reserved for @native later (that's not much, but well...), and remaining gives some buffer for firmware growth and user usage.

pfalcon, 2016-08-05:
This is intended to be a fast-track RFC.

dpgeorge, 2016-08-05:
Since we are already at 514k firmware size I'd suggest we need more than (512+64)k=0x90000 total space.  If we accept that the firmware is >512k that means the ESP module needs at least 1mb flash, so using, say, (512+128)=640k=0xa0000 of that for the firmware would still leave 384k for the filesystem, which is a decent amount (and modules with 4mb flash have a whole lot more).

Note that just compiling in @native support will add roughly 24k to the firmware size for the native emitter (tested by compiling esp8266 port with MICROPY_EMIT_THUMB and MICROPY_EMIT_INLINE_THUMB enabled).

pfalcon, 2016-08-05:
> 384k for the filesystem, which is a decent amount

Not that much, I'd say. It doesn't seem good idea to reserve too much for something which doesn't yet exist. So, if 0x90000 doesn't seem to will have helped with to be developed features, let's leave it at 0x89000 for now, and just make 5-sector change for SDK2.0.0 compat.

dpgeorge, 2016-08-06:
SDK2 compatibility is the important thing, so at least change that.

An alternative scheme for the FS is to always use as much as possible (rounded to 4k) based on the firmware size.  This would be auto-detected on boot up.  It would mean that flashing the firmware would require recreating the FS.

pfalcon, 2016-08-06:
>  It would mean that flashing the firmware would require recreating the FS.

Yes, and that's what we wanted to avoid from the beginning.

Anyway, decided, just reserve an extra sector for 2.0.0. (Actual switch to 2.0.0 is grey area, I have a feeling firmware with it no longer can be flashed over clean ROM (well, their wording implies that, and it kinda feels like that, need clean-room testing)).

pfalcon, 2016-08-06:
Done in ca59f5f208f17620a0ac2fc77e3026bed0d9fe2c.

pfalcon, 2016-08-09:
OK, with addition of umqtt.robust, 0x89000 doesn't fit the image, so I do bump filesystem start to 0x90000.

pfalcon, 2016-08-09:
Done, and went into 1.8.3

closed 2016-08-09 by pfalcon