micropython/micropython #2295: Preparing to 1.8.3 release

pfalcon, 2016-08-05:
* https://github.com/micropython/micropython/pull/2286
* https://github.com/micropython/micropython/issues/2294

dpgeorge, 2016-08-06:
What about #2293?

pfalcon, 2016-08-06:
Commented on #2293, imho we've covered that for releases long ago.

mcauser, 2016-08-07:
Has there been any further progress with the HSPI module since #2170 and #2172 ?

dpgeorge, 2016-08-08:
@pfalcon I'm ready to tag v1.8.3, but maybe wait for WebREPL GET_VER stuff to be finished?

pfalcon, 2016-08-08:
> I'm ready to tag v1.8.3, but maybe wait for WebREPL GET_VER stuff to be finished?

It should be ready from MicroPython side. Well, it doesn't follow request/response structure of other commands, and due to its nature, will have to stay that way forever. So, if there's any criticism, please share it, otherwise, let's tag.

dpgeorge, 2016-08-08:
> Well, it doesn't follow request/response structure of other commands, and due to its nature, will have to stay that way forever.

GET_VER could be changed to be consistent with the others by returning: {'W', 'B', code_lo, code_hi}, and then "code" could be the number of bytes to read for the version.  That way we can change the format of version in the future to be different from 3 bytes (client would check that it is compatible with the value returned in code).  For example we may want to add a sub version number like the git hash or number of git commits since the last tag.

pfalcon, 2016-08-08:
Yes, I could be, but should it be? The idea is that 'W', 'B', etc. stuff may change, but GET_VER will stay. That's why it's made dead simple. The only further "improvement" is to reduce it down to a single byte, a protocol version. But then it will be only harder to match uPy and other component version to protocol.

> For example we may want to add a sub version number like the git hash or number of git commits since the last tag.

I doubt somebody will manage it at such detail. The idea is to protect users (== running a release version).

dpgeorge, 2016-08-09:
> The only further "improvement" is to reduce it down to a single byte, a protocol version. But then it will be only harder to match uPy and other component version to protocol.

I think it's necessary to do that (have a protocol version, not uPy version).  Consider the following: in a month we go to v1.8.4 but make no changes to the webrepl protocol.  Webrepl client stays the same.  But when webrepl client is used to connect to a v1.8.4 board it sees a newer version of the protocol and cannot communicate because it doesn't know if anything changed between 1.8.3 and 1.8.4.  Conversely, if we want to make changes to the webrepl protocol and want to propagate them ASAP, then we're forced to bump the uPy version number.

I propose to have a 2 byte protocol version number.  First byte is the major number, and the meaning of the second byte is dependend on the major number (it can be 0 for now an unused, but gives room to grow in the future, eg with a capabilities byte).

pfalcon, 2016-08-09:
> Webrepl client stays the same. But when webrepl client is used to connect to a v1.8.4 board it sees a newer version of the protocol and cannot communicate because it doesn't know if anything changed between 1.8.3 and 1.8.4.

If it stays the same, it "knows" that nothing important changed, as otherwise we'd modify it to issue warning/error on old version. The idea is to have minimal required webrepl server version check. (Then possibly for additional features, individual checks.)

> Conversely, if we want to make changes to the webrepl protocol and want to propagate them ASAP, then we're forced to bump the uPy version number.

As long as we release uPy versions monthly, we can (and do) wait until next release, and I don't see any problems with that.

So, I don't see any big improvements switching one for another, but feel free to do that, if you want. My idea, is that it's better to report users "File transfers support requires MicroPython of at least 1.8.3 version" than "File transfers support requires MicroPython with WebREPL protocol of at least 123.25 version".

dpgeorge, 2016-08-09:
> If it stays the same, it "knows" that nothing important changed, as otherwise we'd modify it to issue warning/error on old version.

But isn't the point to also protect users using an old (local) version of webrepl client, which we have no way of changing if the protocol changes?  Eg I'm unknowningly using an old webrepl that supports up to 1.8.3 and it connects to a 1.9.6 board... how does it know if it uses a compatible protocol?

pfalcon, 2016-08-09:
> But isn't the point to also protect users using an old (local) version of webrepl client, which we have no way of changing if the protocol changes?

Well, that's harder usecase. The only way to deal with that is indeed to require detailed protocol version management. I'm approaching much simpler usecase - people having older firmware accessing newer online client. That should cover 90% of issue. I'd say it's fair to expect that if people host webrepl locally, they also should be ready to update it (well, we should make that explicit it docs).

pfalcon, 2016-08-09:
Anyway, as I said, if you feel like making that change, please go for it. I over weekend have done much more than I expected to do, and feel that this part is "good enough" ;-).

pfalcon, 2016-08-09:
1.8.3 has been tagged: https://github.com/micropython/micropython/releases/tag/v1.8.3

closed 2016-08-09 by pfalcon