micropython/micropython #2296: upip raises an errno 97 trying to connect to pipy on linux

rguillon, 2016-08-06:
I'm tying to use upip and gets the following error, I've just uncommented some logs:

~/esp/micropython/unix $ micropython -m upip install foobar
Installing to: /home/renaud/.micropython/lib/
Address infos: [(10, 1, 6, None, bytearray(b'\n\x00\x01\xbb\x00\x00\x00\x00*\x04NB\x00\x03\x00\x00\x00\x00\x00\x00\x00\x00\x02#\x00\x00\x00\x00')), (10, 2, 17, None, bytearray(b'\n\x00\x01\xbb\x00\x00\x00\x00*\x04NB\x00\x03\x00\x00\x00\x00\x00\x00\x00\x00\x02#\x00\x00\x00\x00')), (10, 3, 0, None, bytearray(b'\n\x00\x01\xbb\x00\x00\x00\x00*\x04NB\x00\x03\x00\x00\x00\x00\x00\x00\x00\x00\x02#\x00\x00\x00\x00')), (2, 1, 6, None, bytearray(b'\x02\x00\x01\xbb\x97e\x0c\xdf\x00\x00\x00\x00\x00\x00\x00\x00')), (2, 2, 17, None, bytearray(b'\x02\x00\x01\xbb\x97e\x0c\xdf\x00\x00\x00\x00\x00\x00\x00\x00')), (2, 3, 0, None, bytearray(b'\x02\x00\x01\xbb\x97e\x0c\xdf\x00\x00\x00\x00\x00\x00\x00\x00'))]
Connect address: bytearray(b'\n\x00\x01\xbb\x00\x00\x00\x00*\x04NB\x00\x03\x00\x00\x00\x00\x00\x00\x00\x00\x02#\x00\x00\x00\x00')
Traceback (most recent call last):
  File "upip.py", line 299, in <module>
  File "upip.py", line 294, in main
  File "upip.py", line 284, in main
  File "upip.py", line 187, in install_pkg
  File "upip.py", line 165, in get_pkg_metadata
  File "upip.py", line 125, in download
OSError: 97

I'm using Linux Mint 18, any idea on how I could fix that?

pfalcon, 2016-08-06:
In your case, ipv6 addresses are returned first in preference of ipv4, then something goes wrong when actually using it. I'll investigate in detail when time permits.

pfalcon, 2016-09-12:
This should be fixed by https://github.com/micropython/micropython-lib/pull/101 (not yet propagated to upy builtin version).