micropython/micropython #2297: Pin.irq() signature is not Pythonic

pfalcon, 2016-08-07:
I figured that Pin object has .irq() method via REPL auto-completion. I expect following to work (as in - produce visible effect on IRQ):

~~~
b.irq(lambda: print("here"))
~~~

deshipu, 2016-08-08:
What do you mean it's "not pythonic"? It's perfectly fine, albeit completely different from what you wrote, because the concept is much more complex than what you anticipated.

The documentation at http://micropython.org/resources/docs/en/latest/esp8266/library/machine.Pin.html#machine.Pin.irq tells you exactly how to use it, maybe you should refer to it.

dpgeorge, 2016-08-26:
I don't understand how this issue is to be fixed... the API is as it is, do you have an alternative suggestion?

pfalcon, 2016-08-26:
Sure, handler arg should be accepted as positional, trigger should be optional and default to "all" (both edges in this case, whenever possible).

dpgeorge, 2016-08-26:
> trigger should be optional and default to "all" (both edges in this case, whenever possible).

That doesn't really work for other ports where the trigger can be other things, like the logic level.  Also, this signature should be similar for other peripherals like UART (which has lots of triggers as well).

The trigger is more important than the handler (and should be first in the argument list) because the irq can do more than just call the handler (eg it can wake a device from sleep).  If you're going to enable an irq handler then you should at the very least specify what you're triggering on.  That's the rationale.

pfalcon, 2016-08-26:
> That doesn't really work for other ports where the trigger can be other things, like the logic level. Also, this signature should be similar for other peripherals like UART (which has lots of triggers as well).

I think there may be confusion between IRQ edge/level setup and different IRQs per block. So, an external interrupt commonly can be configured as happening on rising edge, falling edge, both, or on level (usually just one level, though I imagine can be configurable). In this case, it's property of IRQ, it's a "type", not "event". There're no separate vectors for each of them, just one.

This is different from UART interrupts, where there can be a separate TX and RX vectors.

> If you're going to enable an irq handler then you should at the very least specify what you're triggering on.

Where there're "filters" (like edge/level type), default should be "all" (or other sensible value).

dpgeorge, 2016-08-27:
> I think there may be confusion between IRQ edge/level setup and different IRQs per block

The point is that different MCUs multiplex IRQs in different ways.  Eg for UART rx/tx, some MCUs might have them at different vectors, and some may put them together with a flag set to distinguish which actually triggered.

The idea behind the HW API was to provide a common ground for these differences by having one "IRQ" per peripheral (eg per pin, per UART) and group together all "triggers" for that peripheral (eg the UART will have rx and tx go to the same Python callback, with a flag set to tell which one occurred).

dpgeorge, 2016-09-20:
Let's try to sort this issue out.  The current irq method signature is:
```python
peripheral.irq(*, trigger=None, priority=1, handler=None, wake=None)
```
The idea is that there is one irq per peripheral instance (eg per Pin, per UART, per I2C, etc; I don't know about virtual peripherals like bit-bang I2C).  That irq is disabled by default.  If you want to enable the irq and set the handler you write:
```python
irq = peripheral.irq(handler=lambda p:...)
```
Then the `irq` object that is returned has some methods to enable/disable it, check the interrupt flags, and do a software trigger of the handler.  You can get the irq object without changing any of the config values by writing:
```python
irq = peripheral.irq()
```
For more details the HW API page has a near-complete/correct description: https://github.com/micropython/micropython/wiki/Hardware-API#irqs

This irq() method acts like an config function (set handler, trigger, etc) and an accessor (get irq object) at the same time, which might be a bit confusing.  The rationale to do it this way was so that you can set a function to be called by the irq on one line, with one method call.  The alternative would be to have periph.irq() just return the irq object, and then all configuration is done by calling methods on that irq object.

If we want to add soft irqs we could add an argument, something like:
```python
periph.irq(handler, trigger, priority, wake, hard=False)
```
But that might be putting too much functionality in one method.  Now there are soft and hard irq objects (or are they the same object just with two handlers?).  A simpler alternative could be to just provide accessor methods like:
```python
h_irq = periph.hard_irq() # get hard irq object
h_irq.handler(lambda p:...) # set handler
s_irq = periph.soft_irq() # get soft irq object
s_irq.handler(lambad p:..., priority=4) # set handler and priority
```

pfalcon, 2016-09-23:
>  The current irq method signature is:

~~~~
peripheral.irq(*, trigger=None, priority=1, handler=None, wake=None)
~~~~

So, I propose to make this:

~~~~
peripheral.irq(handler=None, trigger=None, *, priority=1, wake=None)
~~~~

This should be even backward-compatible.

> This irq() method acts like an config function (set handler, trigger, etc) and an accessor (get irq object) at the same time, which might be a bit confusing.

Maybe it's indeed is, a bit.

> The rationale to do it this way was so that you can set a function to be called by the irq on one line, with one method call. 

That's how constructors in machine.* work, so hopefully once someone reads the  whole docs, it makes sense.

> The alternative would be to have periph.irq() just return the irq object, and then all configuration is done by calling methods on that irq object.

Well, that would be very verbose, and movement in opposite direction wrt to my complaint: I'm calling to make a (pretty tiny by the existing state) special case for quick&easy handler setting, you propose how to make it "harder" in every case. The only question I would actually have - is this "irq" object static as we try to do with hardware objects, or allocated in heap?

~~~~
h_irq = periph.hard_irq() # get hard irq object
h_irq.handler(lambda p:...) # set handler
s_irq = periph.soft_irq() # get soft irq object
s_irq.handler(lambad p:..., priority=4) # set handler and priority
~~~~

AFAIU, hard and irq handlers are exclusive - if you set soft irq handler, underlyingly, hard irq handler is set, which queues this irq to be processed as an event later, at safe time.

dpgeorge, 2016-09-26:
> So, I propose to make this:
> 
> peripheral.irq(handler=None, trigger=None, *, priority=1, wake=None)
> 
> This should be even backward-compatible.

Yes, okay, let's use this.  Although if the trigger is not specified it'll default to something sensible (like IRQ_RISING | IRQ_FALLING), not `None`.

> The only question I would actually have - is this "irq" object static as we try to do with hardware objects, or allocated in heap?

Yes it should be statically allocated.  And it's almost a requirement because in a hard irq handler you need to get the irq object using `periph.irq()`.

> AFAIU, hard and irq handlers are exclusive - if you set soft irq handler, underlyingly, hard irq handler is set, which queues this irq to be processed as an event later, at safe time.

In my mind hard and soft were going to be independently settable.  But you're right that it's simpler to have them exclusive.  And if you need to use hard irq and soft together then you can just configure the hard handler to raise a soft irq yourself.

In that case, would the API for irq handlers be:
```
peripheral.irq(handler=None, trigger=None, *, priority=1, wake=None, hard=False)
```
?

dpgeorge, 2017-06-14:
Pin.irq() is now documented as allowing to pass just a callback for the handler, and it will work "as expected".

closed 2017-06-14 by dpgeorge