micropython/micropython #2298: mpy-cross fails to compile

ryannathans, 2016-08-08:
Building mpy-cross results in the following

```
mkdir -p build/genhdr
python ../py/makeversionhdr.py build/genhdr/mpversion.h
Generating build/genhdr/mpversion.h
GEN build/genhdr/qstr.i.last
if [ "../py/mpstate.c ../py/malloc.c ../py/gc.c ../py/qstr.c ../py/vstr.c ../py/mpprint.c ../py/unicode.c ../py/mpz.c ../py/lexer.c ../py/lexerstr.c ../py/lexerunix.c ../py/parse.c ../py/scope.c ../py/compile.c ../py/emitcommon.c ../py/emitbc.c ../py/asmx64.c ../py/asmx86.c ../py/asmthumb.c ../py/emitinlinethumb.c ../py/asmarm.c ../py/formatfloat.c ../py/parsenumbase.c ../py/parsenum.c ../py/emitglue.c ../py/runtime.c ../py/runtime_utils.c ../py/nativeglue.c ../py/stackctrl.c ../py/argcheck.c ../py/warning.c ../py/map.c ../py/obj.c ../py/objarray.c ../py/objattrtuple.c ../py/objbool.c ../py/objboundmeth.c ../py/objcell.c ../py/objclosure.c ../py/objcomplex.c ../py/objdict.c ../py/objenumerate.c ../py/objexcept.c ../py/objfilter.c ../py/objfloat.c ../py/objfun.c ../py/objgenerator.c ../py/objgetitemiter.c ../py/objint.c ../py/objint_longlong.c ../py/objint_mpz.c ../py/objlist.c ../py/objmap.c ../py/objmodule.c ../py/objobject.c ../py/objpolyiter.c ../py/objproperty.c ../py/objnone.c ../py/objnamedtuple.c ../py/objrange.c ../py/objreversed.c ../py/objset.c ../py/objsingleton.c ../py/objslice.c ../py/objstr.c ../py/objstrunicode.c ../py/objstringio.c ../py/objtuple.c ../py/objtype.c ../py/objzip.c ../py/opmethods.c ../py/sequence.c ../py/stream.c ../py/binary.c ../py/builtinimport.c ../py/builtinevex.c ../py/modarray.c ../py/modbuiltins.c ../py/modcollections.c ../py/modgc.c ../py/modio.c ../py/modmath.c ../py/modcmath.c ../py/modmicropython.c ../py/modstruct.c ../py/modsys.c ../py/moduerrno.c ../py/modthread.c ../py/vm.c ../py/bc.c ../py/showbc.c ../py/repl.c ../py/smallint.c ../py/frozenmod.c ../py/../extmod/moductypes.c ../py/../extmod/modujson.c ../py/../extmod/modure.c ../py/../extmod/moduzlib.c ../py/../extmod/moduheapq.c ../py/../extmod/moduhashlib.c ../py/../extmod/modubinascii.c ../py/../extmod/virtpin.c ../py/../extmod/machine_mem.c ../py/../extmod/machine_pinbase.c ../py/../extmod/machine_pulse.c ../py/../extmod/machine_i2c.c ../py/../extmod/modussl_axtls.c ../py/../extmod/modurandom.c ../py/../extmod/modwebsocket.c ../py/../extmod/modwebrepl.c ../py/../extmod/modframebuf.c ../py/../extmod/fsusermount.c ../py/../extmod/vfs_fat.c ../py/../extmod/vfs_fat_ffconf.c ../py/../extmod/vfs_fat_diskio.c ../py/../extmod/vfs_fat_file.c ../py/../extmod/vfs_fat_lexer.c ../py/../extmod/vfs_fat_misc.c ../py/../extmod/moduos_dupterm.c ../py/../lib/embed/abort_.c ../py/emitnative.c" = "" ]; then \
    echo "QSTR Looks like -B used, trying to emulate"; \
    gcc -E -DNO_QSTR -DN_X64 -DN_X86 -DN_THUMB -DN_ARM -Ibuild/tmp -I../lib -I../lib/netutils -I. -I.. -Ibuild -Wall -Werror -Wpointer-arith -Wuninitialized -ansi -std=gnu99  -Os   -fdata-sections -ffunction-sections -fno-asynchronous-unwind-tables ../py/mpstate.c ../py/malloc.c ../py/gc.c ../py/qstr.c ../py/vstr.c ../py/mpprint.c ../py/unicode.c ../py/mpz.c ../py/lexer.c ../py/lexerstr.c ../py/lexerunix.c ../py/parse.c ../py/scope.c ../py/compile.c ../py/emitcommon.c ../py/emitbc.c ../py/asmx64.c ../py/asmx86.c ../py/asmthumb.c ../py/emitinlinethumb.c ../py/asmarm.c ../py/formatfloat.c ../py/parsenumbase.c ../py/parsenum.c ../py/emitglue.c ../py/runtime.c ../py/runtime_utils.c ../py/nativeglue.c ../py/stackctrl.c ../py/argcheck.c ../py/warning.c ../py/map.c ../py/obj.c ../py/objarray.c ../py/objattrtuple.c ../py/objbool.c ../py/objboundmeth.c ../py/objcell.c ../py/objclosure.c ../py/objcomplex.c ../py/objdict.c ../py/objenumerate.c ../py/objexcept.c ../py/objfilter.c ../py/objfloat.c ../py/objfun.c ../py/objgenerator.c ../py/objgetitemiter.c ../py/objint.c ../py/objint_longlong.c ../py/objint_mpz.c ../py/objlist.c ../py/objmap.c ../py/objmodule.c ../py/objobject.c ../py/objpolyiter.c ../py/objproperty.c ../py/objnone.c ../py/objnamedtuple.c ../py/objrange.c ../py/objreversed.c ../py/objset.c ../py/objsingleton.c ../py/objslice.c ../py/objstr.c ../py/objstrunicode.c ../py/objstringio.c ../py/objtuple.c ../py/objtype.c ../py/objzip.c ../py/opmethods.c ../py/sequence.c ../py/stream.c ../py/binary.c ../py/builtinimport.c ../py/builtinevex.c ../py/modarray.c ../py/modbuiltins.c ../py/modcollections.c ../py/modgc.c ../py/modio.c ../py/modmath.c ../py/modcmath.c ../py/modmicropython.c ../py/modstruct.c ../py/modsys.c ../py/moduerrno.c ../py/modthread.c ../py/vm.c ../py/bc.c ../py/showbc.c ../py/repl.c ../py/smallint.c ../py/frozenmod.c ../py/../extmod/moductypes.c ../py/../extmod/modujson.c ../py/../extmod/modure.c ../py/../extmod/moduzlib.c ../py/../extmod/moduheapq.c ../py/../extmod/moduhashlib.c ../py/../extmod/modubinascii.c ../py/../extmod/virtpin.c ../py/../extmod/machine_mem.c ../py/../extmod/machine_pinbase.c ../py/../extmod/machine_pulse.c ../py/../extmod/machine_i2c.c ../py/../extmod/modussl_axtls.c ../py/../extmod/modurandom.c ../py/../extmod/modwebsocket.c ../py/../extmod/modwebrepl.c ../py/../extmod/modframebuf.c ../py/../extmod/fsusermount.c ../py/../extmod/vfs_fat.c ../py/../extmod/vfs_fat_ffconf.c ../py/../extmod/vfs_fat_diskio.c ../py/../extmod/vfs_fat_file.c ../py/../extmod/vfs_fat_lexer.c ../py/../extmod/vfs_fat_misc.c ../py/../extmod/moduos_dupterm.c ../py/../lib/embed/abort_.c ../py/emitnative.c >build/genhdr/qstr.i.last; \
else \
    gcc -E -DNO_QSTR -DN_X64 -DN_X86 -DN_THUMB -DN_ARM -Ibuild/tmp -I../lib -I../lib/netutils -I. -I.. -Ibuild -Wall -Werror -Wpointer-arith -Wuninitialized -ansi -std=gnu99  -Os   -fdata-sections -ffunction-sections -fno-asynchronous-unwind-tables ../py/mpstate.c ../py/malloc.c ../py/gc.c ../py/qstr.c ../py/vstr.c ../py/mpprint.c ../py/unicode.c ../py/mpz.c ../py/lexer.c ../py/lexerstr.c ../py/lexerunix.c ../py/parse.c ../py/scope.c ../py/compile.c ../py/emitcommon.c ../py/emitbc.c ../py/asmx64.c ../py/asmx86.c ../py/asmthumb.c ../py/emitinlinethumb.c ../py/asmarm.c ../py/formatfloat.c ../py/parsenumbase.c ../py/parsenum.c ../py/emitglue.c ../py/runtime.c ../py/runtime_utils.c ../py/nativeglue.c ../py/stackctrl.c ../py/argcheck.c ../py/warning.c ../py/map.c ../py/obj.c ../py/objarray.c ../py/objattrtuple.c ../py/objbool.c ../py/objboundmeth.c ../py/objcell.c ../py/objclosure.c ../py/objcomplex.c ../py/objdict.c ../py/objenumerate.c ../py/objexcept.c ../py/objfilter.c ../py/objfloat.c ../py/objfun.c ../py/objgenerator.c ../py/objgetitemiter.c ../py/objint.c ../py/objint_longlong.c ../py/objint_mpz.c ../py/objlist.c ../py/objmap.c ../py/objmodule.c ../py/objobject.c ../py/objpolyiter.c ../py/objproperty.c ../py/objnone.c ../py/objnamedtuple.c ../py/objrange.c ../py/objreversed.c ../py/objset.c ../py/objsingleton.c ../py/objslice.c ../py/objstr.c ../py/objstrunicode.c ../py/objstringio.c ../py/objtuple.c ../py/objtype.c ../py/objzip.c ../py/opmethods.c ../py/sequence.c ../py/stream.c ../py/binary.c ../py/builtinimport.c ../py/builtinevex.c ../py/modarray.c ../py/modbuiltins.c ../py/modcollections.c ../py/modgc.c ../py/modio.c ../py/modmath.c ../py/modcmath.c ../py/modmicropython.c ../py/modstruct.c ../py/modsys.c ../py/moduerrno.c ../py/modthread.c ../py/vm.c ../py/bc.c ../py/showbc.c ../py/repl.c ../py/smallint.c ../py/frozenmod.c ../py/../extmod/moductypes.c ../py/../extmod/modujson.c ../py/../extmod/modure.c ../py/../extmod/moduzlib.c ../py/../extmod/moduheapq.c ../py/../extmod/moduhashlib.c ../py/../extmod/modubinascii.c ../py/../extmod/virtpin.c ../py/../extmod/machine_mem.c ../py/../extmod/machine_pinbase.c ../py/../extmod/machine_pulse.c ../py/../extmod/machine_i2c.c ../py/../extmod/modussl_axtls.c ../py/../extmod/modurandom.c ../py/../extmod/modwebsocket.c ../py/../extmod/modwebrepl.c ../py/../extmod/modframebuf.c ../py/../extmod/fsusermount.c ../py/../extmod/vfs_fat.c ../py/../extmod/vfs_fat_ffconf.c ../py/../extmod/vfs_fat_diskio.c ../py/../extmod/vfs_fat_file.c ../py/../extmod/vfs_fat_lexer.c ../py/../extmod/vfs_fat_misc.c ../py/../extmod/moduos_dupterm.c ../py/../lib/embed/abort_.c ../py/emitnative.c >build/genhdr/qstr.i.last; \
fi
GEN build/genhdr/qstr.split
python ../py/makeqstrdefs.py split build/genhdr/qstr.i.last build/genhdr/qstr build/genhdr/qstrdefs.collected.h
touch build/genhdr/qstr.split
GEN build/genhdr/qstrdefs.collected.h
python ../py/makeqstrdefs.py cat build/genhdr/qstr.i.last build/genhdr/qstr build/genhdr/qstrdefs.collected.h
QSTR updated
GEN build/genhdr/qstrdefs.generated.h
[... cut ...]
gcc -I../lib -I../lib/netutils -I. -I.. -Ibuild -Wall -Werror -Wpointer-arith -Wuninitialized -ansi -std=gnu99  -Os   -fdata-sections -ffunction-sections -fno-asynchronous-unwind-tables -c -MD -o build/windows/fmode.o ../windows/fmode.c

../windows/fmode.c: In function ‘set_fmode_impl’:
../windows/fmode.c:36:5: error: ‘_fmode’ undeclared (first use in this function)
     _fmode = mode;
     ^
../windows/fmode.c:36:5: note: each undeclared identifier is reported only once for each function it appears in
make: *** [../py/mkrules.mk:47: build/windows/fmode.o] Error 1
```

dpgeorge, 2016-08-08:
What OS are you building on?

ryannathans, 2016-08-08:
@dpgeorge Windows 10, in a cygwin/bash terminal

dpgeorge, 2016-08-08:
This is one for @stinos .

robert-hh, 2016-08-08:
Windows10/bash mimics Linux. So you should be able to build the linux version. If you want to build the Windows version, use plain Windows cmd and for instance the mingw.

ryannathans, 2016-08-08:
@robert-hh I'm not quite sure what you're getting at. I'm running "make" in the mpy-cross directory and encountering these errors

stinos, 2016-08-08:
> use plain Windows cmd and for instance the mingw

or cygwin's bash shell and mingw-w64 (plain mingw isn't supported anymore for uPy)

which is what I did, and I can't reproduce the issue.
So I need more info: what is your exact command line, and what is the compiler version?

ryannathans, 2016-08-08:
> It looks like your intention was not clear to me either.
> mpy-cross is intended to be used on the host machine, which is, as I understand, windows10/ bash shell. Within the bash shell, you can 'make' mpy-cross w/o any problems, after installing the required build tools, e.g. using "apt-get install build-essential". If you want to create images for Pyboard or esp8266, you would indeed need the respective cross compilers for that platform.
>
> ---
> https://github.com/micropython/micropython/issues/2298#issuecomment-238290216

Sorry for confusion. Overworked.
Just following my usual build procedure on windows 10.
1. Open cygwin shell in mpy-cross2. Type "make"
Will report versions when I get to work

stinos, 2016-08-10:
Ok got it. fmode.c shouldn't be included at all when using plain gcc on cygwin. As I already thought when committing the previous mpy-cross Makefile patch the detection for when to include fmode.c is broken. I think #2310 should properly fix this. @ryannathans, @dmazzella can you confirm this?

stinos, 2017-06-14:
@dpgeorge this was also fixed by 5b57ae9 so can be closed

dpgeorge, 2017-06-14:
Ok, thanks @stinos!

closed 2017-06-14 by dpgeorge