micropython/micropython #2300: WebREPL has a character limit of 9 in the password?

skorokithakis, 2016-08-08:
Setting a 10-character password fails, but 9 characters work:

~~~
>>>webrepl.start(8266, password="passwordp")
WebREPL daemon started on ws://192.168.4.1:8266
WebREPL daemon started on ws://192.168.0.106:8266
Started webrepl in normal mode
>>> 
WebREPL connection from: ('192.168.0.101', 36296)
# Works, I can connect

>>> webrepl.start(8266, password="passwordpa")
# I can never connect
~~~

I haven't found a limit in the documentation (or any documentation on the WebREPL). Is this as intended?

iSevenDays, 2016-08-08:
Limit is 9 or 10 characters, I found them in source code. It looks like char passoword[9] or [10]

pfalcon, 2016-08-09:
Yes, it's intended to have a limit on password length (unlimited length will never be supported). The exact value is not documented and subject to change without notice. Just as ability to pass password to webrepl.start() is not documented (feature for advanced users). However, if you follow the official way to set webrepl (on first webrepl connection), you won't be allowed to enter too long password.

closed 2016-08-09 by pfalcon

pfalcon, 2016-08-16:
Ok, I gave up and made sure there's no silent operation when setting too long password: d2cc7c720b3c6cd0f48f235a04d2e0bb740dad44 . All this stuff takes bytes. If it'll go that way, soon there will be no useful filesystem on 1MB modules.