micropython/micropython #2303: Micropython PYB soft reboot gc memory problems

iSevenDays, 2016-08-08:
I wanted to have a support for ESP8266 in Pymakr, so I posted information about websockets integration and ESP8266 here https://github.com/danicampora/pymakr-setup-kitchen/issues/1

Then I found that there is a problem with memory, that I can connect to the ESP8266, but I am unable to execute commands and enter REPL mode.

I found that problem is in soft reboot, every time after soft reboot I see the error "OSError: [Errno 12] ENOMEM"

I fixed it with https://github.com/micropython/micropython/pull/2302 , but still, with frequent soft reboots I can get ENOMEM.

To reproduce the error (Mac OS example)

1. Copy pyboard.py from https://gist.github.com/iSevenDays/5440c1f0b7a7dbbe380f5db0fbe56dd9 and replace /Applications/Pymakr.app/Contents/Resources/src/Plugins/UPython/pyboard.py 

2. Start webrepl on ESP8266
import webrepl
webrepl.start()

3. Try to connect to ESP8266 from Pymakr IDE, in Pymakr ide enter the following settings

ws://192.168.76.9:8266 where 192.168.76.9 is the ip of the ESP8266

enter your password

user name is not used

4. See the logs from ESP8266 via Serial connection

Below is the example of such logs.
```
>>> 
MicroPython v1.8.2-88-g4a27ad0-dirty on 2016-08-08; ESP module with ESP8266
Type "help()" for more information.
>>> 
PYB: soft reboot
GC: total: 28160, used: 10480, free: 17680
 No. of 1-blocks: 159, 2-blocks: 31, max blk sz:#6 ets_task(401003a4, 3, 3fff6300, 4)
Fatal exception 9(LoadStoreAlignmentCause):
epc1=0x4021121a, epc2=0x00000000, epc3=0x00000000, excvaddr=0x0000141e, depc=0x00000000

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 30852, room 16 
tail 4
chksum 0x46
load 0x3ffe8000, len 1064, room 4 
tail 4
chksum 0x3c
load 0x3ffe8430, len 3000, room 4 
tail 4
chksum 0x4a
csum 0x4a
��l��|��rroB���l�c�lb�ܜ��b�l{lllܞ|�2���{Nb���l��p�"�c�ܜ��B��$bd�dܾ���rrn#�������#�"쌜���b�$bd��{�$����l �o�l�l�l`��r�l���l$`s$��"��bc$�c{|�"p���p2"��N�Nn�$l��l쎌�d��d$�����r$�o���"cd쎄b���c#l�"p{$rdr�n����bsb����������pb�#�|�c{c��c��������{2Bl�l��|��rrn#���$���c�"��ܜ���c$d"d��lo�l`��lo�p{$�lܟ|�2���rNb���l����#5 ets_task(401003a4, 3, 3fff6300, 4)
WebREPL daemon started on ws://192.168.4.1:8266
scandone
state: 0 -> 2 (b0)
WebREPL daemon started on ws://0.0.0.0:8266
Started webrepl in normal mode
state: 2 -> 3 (0)
could not open file 'main.py' for reading

MicroPython v1.8.2-88-g4a27ad0-dirty on 2016-08-08; ESP module with ESP8266
Type "help()" for more information.
>>> state: 3 -> 5 (10)
add 0
aid 2
cnt 

connected with Apple, channel 7
dhcp client start...
ip:192.168.76.9,mask:255.255.255.0,gw:192.168.76.1

WebREPL connection from: ('192.168.76.4', 51436)

>>> 
MicroPython v1.8.2-88-g4a27ad0-dirty on 2016-08-08; ESP module with ESP8266
Type "help()" for more information.
>>> 
PYB: soft reboot
GC: total: 28160, used: 10480, free: 17680
 No. of 1-blocks: 159, 2-blocks: 31, max blk sz:#6 ets_task(401003a4, 3, 3fff6300, 4)
Fatal exception 9(LoadStoreAlignmentCause):
epc1=0x4021121a, epc2=0x00000000, epc3=0x00000000, excvaddr=0x0000001e, depc=0x00000000

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 30852, room 16 
tail 4
chksum 0x46
load 0x3ffe8000, len 1064, room 4 
tail 4
chksum 0x3c
load 0x3ffe8430, len 3000, room 4 
tail 4
chksum 0x4a
csum 0x4a
��l��|�����{o"���l�#��"쌜���b�$rl$l��|�����{oB���l܎��b�b쌜���#��cd �$ܾ���rrNb���$����c�c�����B�cl ��r�l�d��l`��s�p��l�d`���n�l���$l`rl���b���cd�br|�bp���pc��n�nn�l$��l��$��l$���l��d �N���bbl�p���cd�cprls$r�o����#{c����>��쏌rb�c��o����cs2b����>�����p"l�l��|��rrnc���d���"��"쌜�܎��#$l#l��lo�$`��lN�p{$�d��|��s{oB���$����#5 ets_task(401003a4, 3, 3fff6300, 4)
WebREPL daemon started on ws://192.168.4.1:8266
scandone
state: 0 -> 2 (b0)
WebREPL daemon started on ws://0.0.0.0:8266
Started webrepl in normal mode
could not open file 'main.py' for reading

MicroPython v1.8.2-88-g4a27ad0-dirty on 2016-08-08; ESP module with ESP8266
Type "help()" for more information.
>>> state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 2
cnt 

connected with Apple, channel 7
dhcp client start...
ip:192.168.76.9,mask:255.255.255.0,gw:192.168.76.1

WebREPL connection from: ('192.168.76.4', 51437)

>>> 
MicroPython v1.8.2-88-g4a27ad0-dirty on 2016-08-08; ESP module with ESP8266
Type "help()" for more information.
>>> 
PYB: soft reboot
GC: total: 28160, used: 10480, free: 17680
 No. of 1-blocks: 159, 2-blocks: 31, max blk sz:#6 ets_task(401003a4, 3, 3fff6300, 4)
TypeError: 'int' object is not callable
Traceback (most recent call last):
  File "boot.py", line 1, in <module>
  File "webrepl.py", line 64, in start
  File "webrepl.py", line 21, in setup_conn
OSError: [Errno 12] ENOMEM
could not open file 'main.py' for reading
MicroPython v1.8.2-88-g4a27ad0-dirty on 2016-08-08; ESP module with ESP8266
Type "help()" for more information.
>>> s$l��|�l�|�l�c|����s���b��no�lNn���Bp��l{lr$p�o��l#o�|�$��"��no�l��l �2Nn�dor���N���lp�n�s�����cn�|l���b��no��l`�on�lnr���or� p�n�r�������bN�|��B�|�n��d �oo�{lNs�������s${����$�`��lܟ|�����rnc���$�#��b�ܜ��B�#$`llܞ|�2���{o"���$��p�B�b�ܜ��c��cd �l��|�2���{o"���$���B��b서���b�$bd��{�l�d��d ��r�l�l�l`��{�l���ll rl��n���Bl�Br|�"p2��"bc��o�oo�ld��l��$��$l����d��s$�o���Bbl�p���"l�bp{$rdr�n����Brc��c쎜������{2B���n����#{c��c쎜������{2Bl�l��|�2���so#���$���c�c�ܜ����bllc$���lo�$`��$n�prl�lܞ|��rrn#���$����#5 ets_task(401003a4, 3, 3fff6300, 4)
WebREPL daemon started on ws://192.168.4.1:8266
scandone
state: 0 -> 2 (b0)
WebREPL daemon started on ws://0.0.0.0:8266
Started webrepl in normal mode
state: 2 -> 3 (0)
could not open file 'main.py' for reading

MicroPython v1.8.2-88-g4a27ad0-dirty on 2016-08-08; ESP module with ESP8266
Type "help()" for more information.
>>> state: 3 -> 5 (10)
add 0
aid 2
cnt 

connected with Apple, channel 7
dhcp client start...
ip:192.168.76.9,mask:255.255.255.0,gw:192.168.76.1

WebREPL connection from: ('192.168.76.4', 51443)

>>> 
MicroPython v1.8.2-88-g4a27ad0-dirty on 2016-08-08; ESP module with ESP8266
Type "help()" for more information.
>>> 
PYB: soft reboot
GC: total: 28160, used: 10480, free: 17680
 No. of 1-blocks: 159, 2-blocks: 31, max blk sz:#6 ets_task(401003a4, 3, 3fff6300, 4)
Fatal exception 9(LoadStoreAlignmentCause):
epc1=0x4021121a, epc2=0x00000000, epc3=0x00000000, excvaddr=0x0000141e, depc=0x00000000

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 30852, room 16 
tail 4
chksum 0x46
load 0x3ffe8000, len 1064, room 4 
tail 4
chksum 0x3c
load 0x3ffe8430, len 3000, room 4 
tail 4
chksum 0x4a
csum 0x4a
��$ܞ���r{o"���l�b�$c䌜��"�$rdl$ܾ���rrnc��l��p�B��"쌜��2�#��#d`�lܞ|��rrnc��l���#�#䌜�����b�$"l��r�l�$��l ��r�l�d�l ��{�$���dl`{l��rl��bcd�br|�`��2�c#"��N�oN�l$��d��l��$l����$��{$�o���Bbl�p���c#$�Bprl{lr�o����|�brb�pB�c����pb�#��o�߀�"r#��#�������pbd�l��|��{rNb���l���pp�"쌜�܎��"l$bd��$n�d`��lo�p{$�$ܾ���rrnc��l����#5 ets_task(401003a4, 3, 3fff6300, 4)
WebREPL daemon started on ws://192.168.4.1:8266
scandone
state: 0 -> 2 (b0)
WebREPL daemon started on ws://0.0.0.0:8266
Started webrepl in normal mode
state: 2 -> 3 (0)
could not open file 'main.py' for reading

MicroPython v1.8.2-88-g4a27ad0-dirty on 2016-08-08; ESP module with ESP8266
Type "help()" for more information.
>>> state: 3 -> 5 (10)
add 0
aid 2
cnt 

connected with Apple, channel 7
dhcp client start...
ip:192.168.76.9,mask:255.255.255.0,gw:192.168.76.1
```

pfalcon, 2016-08-12:
Duplicate of https://github.com/micropython/micropython/issues/1896

dpgeorge, 2017-06-14:
As mentioned above this is a duplicate, so closing.

closed 2017-06-14 by dpgeorge