micropython/micropython #2304: Floating operations fails from frozen modules (esp8266)

rguillon, 2016-08-08:
If I run: 
`print(2.0 + 1.0)`

from a frozen modules I get the following error:
`
TypeError: can't convert float to float
`

dpgeorge, 2016-08-09:
Yes this is a known limitation of frozen modules for ports that use 30-bit float stuffing.

peterhinch, 2016-08-09:
It would be good to know what type of code provokes this. I have successfully frozen a module which uses a minimal amount of FP, but freezing another module which calls it provoked that error. Any pointers on workrounds would be welcome.

rguillon, 2016-08-09:
Thanks, 

For the record, the topic disusing the subject: 

https://forum.micropython.org/viewtopic.php?f=2&t=2213&p=12577&hilit=compiler#p12577

dpgeorge, 2016-08-10:
Should now be fixed by 72ae3c72c7341656bcc610a5671e5cc23f7cf280.

rguillon, 2016-08-10:
Confirmed, thanks!

closed 2016-08-10 by rguillon