micropython/micropython #2307: Feature suggestion; WebREPL launch code

cefn, 2016-08-09:
To supply pre-configured modules for learners, it would be really useful to be able to have the WebREPL come up with a series of lines passed to it which it will execute. 

That way, when a learner connects to the WebREPL, they find it already running, with a pre-defined set of symbols available, and can hit CTRL+C to kill the last line or block of code, before running their own in the same context that the original code line or block was running, prepared by the author of the project.

To give you a worked example, this would be ideal for e.g. a light-up badges project I'm putting together for Light-Up Lancaster and the Morecambe Lantern Festival here @ShrimpingIt based on Neopixel arrays. 

Out of the box, a project build will do something meaningful (e.g. set up and run a __rainbow()__ function to make the lights show up fancy). However, users can connect over wifi and access the WebREPL to configure the behaviour of the light show interactively hitting CTRL+C then issuing their own commands.

For example, __rainbow()__, __pingpong()__, __strobe()__ etc. could all be available as bound symbols in the REPL context provided. The learner could equally manipulate a provided __np__ symbol with index assignments followed by a call to __np.write()__.

Learner documentation would tell them what items were available, or they could interrogate with dir(), or use repl() or pprint on globals().

Is this already feasible by getting access to the correct stream handle (e.g. throwing lines directly at the webrepl after start() by going under the hood of the socket byte stream)? I got a bit lost in the webrepl.py event passing via _webrepl._webrepl(ws), and couldn't figure from extmod/modwebrepl.c or from any websocket library where I could intervene from python if this is in fact feasible.

If there is no way to provide these lines, is it possible to extend the webrepl.start() signature with an optional ___script___ argument, which should contain properly indented python.

Alternatively, would this justify monitoring for a _repl.py_ path in the filesystem (alongside _main.py_ and _boot.py_) which the webrepl.py script would interrogate on launch to see if there were lines to be run. This would mean that users could easily change the startup behaviour by re-uploading a file, just like for boot and main, and could easily switch from something run interactively to something run without network wifi or serial by moving a file from its location in repl.py to main.py.

cefn, 2016-08-10:
@pfalcon to achieve this feature (without agreeing any repl.py filesystem convention), is there a way I can call one of the methods in https://github.com/micropython/micropython/blob/master/extmod/modwebrepl.c through the _webrepl module to inject a string or buffer of bytes as if they were sent by the remote client (possibly without there being an actual remote client, yet)?

This would be taking place on boot, so I don't anticipate problems from these bytes interleaving in some way with real bytes from an actual WebREPL client, and this is a price I'm willing to pay anyway.

The result should be that the bytes passed are interpreted as if they were sent by the Serial or Websocket 'tty'. I imagine prefixing them with the Unicode control characters for CTRL+E and terminating with CTRL+D would be a good idea so auto-indent logic doesn't mess with the code.

I hope I have traced the proper files for this job, although I don't yet understand the mechanism by which the _webrepl module is backed by the extmod/modwebrepl.c file if at all.

deshipu, 2016-08-10:
Just put your code in "boot.py" or "main.py" and you will have all the names available in the repl.

cefn, 2016-08-10:
@deshipu seems impossible to access the main.py script context (and CTRL+C the last operation executed) via the WebREPL, as long as main.py fails to return (i.e. if you leave it running anything, after having defined the symbols).

The MicroPython-xxxxxx SSID is available and accepts connections. The WebREPL connection process begins, but no password can be negotiated as if the password accept operation is blocked waiting on something else.

The main.py which creates that issue looks like this...

```
def greet():
    from time import sleep
    print('Hello World')
    sleep(2)

while True:
    greet()
```

However, it turns out that the approach you describe IS fundamentally correct. Turns out I can access the normal Serial REPL, then issue CTRL+C to cancel the while loop. It just doesn't work via the WebREPL as no connection can be negotiated, until a CTRL+C is issued via the Serial REPL. After CTRL+C cancels it via Serial, reloading the WebREPL page, and making a second attempt to access the WebREPL is successful.

Of course, this is essentially no help for our application as it requires either that you have a Serial REPL set up, or that you have no behaviour running, which means the WebREPL cannot be used to 'override' and take control of an existing behaviour deployed to the device.

If I comment out the while loop, then the WebREPL succeeds to connect, and the greet() symbol is available to the REPL just as you promised.

If I could actually connect to the WebREPL in the same way as I can with the Serial REPL this would definitely meet my needs. However, for now, it doesn't seem to function.

cefn, 2016-08-10:
Attempted various ways of removing the password prompt, but it seems impossible without rewriting webrepl, even if I explicitly upload a port_config.py like this...

```
WEBREPL_PASS = None
```

I also don't know if this will make any difference, the WebREPL 'tty' might be blocked anyway as long as main.py doesn't return - not sure if it's to do with the password handshake anyway.

cefn, 2016-08-14:
Looks like the issue I was hitting was resolved July 22nd and I hadn't updated my local master, yet. Now working fine.

closed 2016-08-14 by cefn