micropython/micropython #2309: ESP8266 build problem

peterhinch, 2016-08-10:
With source pulled this morning:
```
[successful lines deleted]
CC ../lib/berkeley-db-1.xx/btree/bt_delete.c
CC ../lib/berkeley-db-1.xx/btree/bt_get.c
In file included from ../lib/berkeley-db-1.xx/PORT/include/../../btree/btree.h:44:0,
                 from ../extmod/modbtree.c:40:
../lib/berkeley-db-1.xx/PORT/include/mpool.h:50:2: error: expected specifier-qualifier-list before 'CIRCLEQ_ENTRY'
  CIRCLEQ_ENTRY(_bkt) hq;  /* hash queue */
  ^
../lib/berkeley-db-1.xx/PORT/include/mpool.h:61:2: error: expected specifier-qualifier-list before 'CIRCLEQ_HEAD'
  CIRCLEQ_HEAD(_lqh, _bkt) lqh; /* lru queue head */
  ^
In file included from ../lib/berkeley-db-1.xx/btree/btree.h:44:0,
                 from ../lib/berkeley-db-1.xx/btree/bt_close.c:50:
../lib/berkeley-db-1.xx/PORT/include/mpool.h:50:2: error: expected specifier-qualifier-list before 'CIRCLEQ_ENTRY'
  CIRCLEQ_ENTRY(_bkt) hq;  /* hash queue */
[more error lines deleted]
```

dpgeorge, 2016-08-10:
git submodule update will probably fix this.

peterhinch, 2016-08-10:
It did indeed. Noted for future reference.

closed 2016-08-10 by peterhinch