micropython/micropython #2311: macOS build failures related to stdio.h

ilovezfs, 2016-08-10:
Unless I build with the following diff, I'm seeing build failure for 1.8.3 on macOS 10.11
```
diff --git a/unix/mpconfigport.h b/unix/mpconfigport.h
index c449e73..7b66b78 100644
--- a/unix/mpconfigport.h
+++ b/unix/mpconfigport.h
@@ -306,4 +306,3 @@ void mp_unix_mark_exec(void);
 #endif
 
 // For debugging purposes, make printf() available to any source file.
-#include <stdio.h>
```

Build errors are
```
CC ../lib/utils/printf.c
../lib/utils/printf.c:43:5: error: expected parameter declarator
int vsnprintf(char *str, size_t size, const char *fmt, va_list ap);
    ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.11.sdk/usr/include/secure/_stdio.h:75:40: note: expanded from macro 'vsnprintf'
  __builtin___vsnprintf_chk (str, len, 0, __darwin_obsz(str), format, ap)
                                       ^
../lib/utils/printf.c:43:5: error: expected ')'
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.11.sdk/usr/include/secure/_stdio.h:75:40: note: expanded from macro 'vsnprintf'
  __builtin___vsnprintf_chk (str, len, 0, __darwin_obsz(str), format, ap)
                                       ^
../lib/utils/printf.c:43:5: note: to match this '('
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.11.sdk/usr/include/secure/_stdio.h:75:29: note: expanded from macro 'vsnprintf'
  __builtin___vsnprintf_chk (str, len, 0, __darwin_obsz(str), format, ap)
                            ^
../lib/utils/printf.c:43:5: error: conflicting types for '__builtin___vsnprintf_chk'
int vsnprintf(char *str, size_t size, const char *fmt, va_list ap);
    ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.11.sdk/usr/include/secure/_stdio.h:75:3: note: expanded from macro 'vsnprintf'
  __builtin___vsnprintf_chk (str, len, 0, __darwin_obsz(str), format, ap)
  ^
../lib/utils/printf.c:43:5: note: '__builtin___vsnprintf_chk' is a builtin with type 'int (char *, unsigned long, int, unsigned long, const char *, struct __va_list_tag *)'
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.11.sdk/usr/include/secure/_stdio.h:75:3: note: expanded from macro 'vsnprintf'
  __builtin___vsnprintf_chk (str, len, 0, __darwin_obsz(str), format, ap)
  ^
../lib/utils/printf.c:44:5: error: expected parameter declarator
int snprintf(char *str, size_t size, const char *fmt, ...);
    ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.11.sdk/usr/include/secure/_stdio.h:57:39: note: expanded from macro 'snprintf'
  __builtin___snprintf_chk (str, len, 0, __darwin_obsz(str), __VA_ARGS__)
                                      ^
../lib/utils/printf.c:44:5: error: expected ')'
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.11.sdk/usr/include/secure/_stdio.h:57:39: note: expanded from macro 'snprintf'
  __builtin___snprintf_chk (str, len, 0, __darwin_obsz(str), __VA_ARGS__)
                                      ^
../lib/utils/printf.c:44:5: note: to match this '('
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.11.sdk/usr/include/secure/_stdio.h:57:28: note: expanded from macro 'snprintf'
  __builtin___snprintf_chk (str, len, 0, __darwin_obsz(str), __VA_ARGS__)
                           ^
../lib/utils/printf.c:44:5: error: conflicting types for '__builtin___snprintf_chk'
int snprintf(char *str, size_t size, const char *fmt, ...);
    ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.11.sdk/usr/include/secure/_stdio.h:57:3: note: expanded from macro 'snprintf'
  __builtin___snprintf_chk (str, len, 0, __darwin_obsz(str), __VA_ARGS__)
  ^
../lib/utils/printf.c:44:5: note: '__builtin___snprintf_chk' is a builtin with type 'int (char *, unsigned long, int, unsigned long, const char *, ...)'
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.11.sdk/usr/include/secure/_stdio.h:57:3: note: expanded from macro 'snprintf'
  __builtin___snprintf_chk (str, len, 0, __darwin_obsz(str), __VA_ARGS__)
  ^
../lib/utils/printf.c:108:5: error: expected parameter declarator
int vsnprintf(char *str, size_t size, const char *fmt, va_list ap) {
    ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.11.sdk/usr/include/secure/_stdio.h:75:40: note: expanded from macro 'vsnprintf'
  __builtin___vsnprintf_chk (str, len, 0, __darwin_obsz(str), format, ap)
                                       ^
[... 48 lines cut ...]
    ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.11.sdk/usr/include/secure/_stdio.h:57:3: note: expanded from macro 'snprintf'
  __builtin___snprintf_chk (str, len, 0, __darwin_obsz(str), __VA_ARGS__)
  ^
../lib/utils/printf.c:44:5: note: '__builtin___snprintf_chk' is a builtin with type 'int (char *, unsigned long, int, unsigned long, const char *, ...)'
int snprintf(char *str, size_t size, const char *fmt, ...);
    ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.11.sdk/usr/include/secure/_stdio.h:57:3: note: expanded from macro 'snprintf'
  __builtin___snprintf_chk (str, len, 0, __darwin_obsz(str), __VA_ARGS__)
  ^
../lib/utils/printf.c:123:5: error: definition of builtin function '__builtin___snprintf_chk'
int snprintf(char *str, size_t size, const char *fmt, ...) {
    ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.11.sdk/usr/include/secure/_stdio.h:57:3: note: expanded from macro 'snprintf'
  __builtin___snprintf_chk (str, len, 0, __darwin_obsz(str), __VA_ARGS__)
  ^
../lib/utils/printf.c:125:18: error: use of undeclared identifier 'fmt'
    va_start(ap, fmt);
                 ^
../lib/utils/printf.c:126:36: error: use of undeclared identifier 'fmt'
    int ret = vsnprintf(str, size, fmt, ap);
                                   ^
18 errors generated.
make: *** [build/lib/utils/printf.o] Error 1
```

pfalcon, 2016-08-10:
Please submit diff which disables the include only for macos.

ilovezfs, 2016-08-10:
@pfalcon https://github.com/micropython/micropython/pull/2312.

closed 2016-08-10 by ilovezfs