micropython/micropython #2314: WeMos D1 Mini os.mkfs() fails to execute

cpranzl, 2016-08-11:
No success in executing os.mkfs()
Version: Micropython 1.8.2-9-g805c2b9 on 2016-07-10; ESP module with ESP8266

Repro:
```
>>> import os
>>> os.mkfs()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'module' object has no attribute 'mkfs'
```

deshipu, 2016-08-11:
What makes you think there should be such a function? I didn't find any mention of it in the documentation.

cpranzl, 2016-08-11:
Statet in issue #2044 

> 
> 
> You need to pass the block device and a mount point:
> 
> os.mkfs(blockdevice, '/mnt')
> 
> In the case of the internal flash the block device is pyb.Flash() and you can format it using:
> 
> flash = pyb.Flash()
> os.umount('/flash') # unmount first
> os.mkfs(flash, '/flash')
> os.mount(flash, '/flash') # mount newly formatted filesystem

```
>>> flash = pyb.Flash()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'pyb' is not defined
```

deshipu, 2016-08-11:
That issue is about the PyBoard, not the ESP8266 port.

cpranzl, 2016-08-11:
So 
```
>>> import os
>>> os.listdir()
['boot.py']
```
is the correct behavior?

deshipu, 2016-08-11:
Yes.

closed 2016-08-11 by cpranzl