micropython/micropython #2316: ESP8266 and SSD1306 I2C OLED causing OSError: I2C Bus Error

Connor-F, 2016-08-13:
I'm trying to get a SSD1306 controlled OLED screen working with the ESP8266. I've added the ssd1306 driver and created my custom firmware and flashed it. However when I try to create the display object I get an OSError.
```
>>> display = ssd1306.SSD1306_I2C(128, 64, i2c, 0x3c)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "ssd1306.py", line 108, in __init__
  File "ssd1306.py", line 36, in __init__
  File "ssd1306.py", line 61, in init_display
  File "ssd1306.py", line 113, in write_cmd
OSError: I2C bus error

```

A similar error has been noted here (http://forum.micropython.org/viewtopic.php?t=290&start=10) saying that no ACK is sent from the screen. This has been addressed by u8glib by allowing the display object to be constructed with a special identifier noting that the screen does not send an ACK.

Any ideas to get this working with MicroPython?

dpgeorge, 2016-08-17:
If it's an error with the hardware then the easiest thing is to get new hardware :)

Otherwise, if you want to use a display that doesn't conform to the I2C protocol then you'll need to write your own I2C driver.  It's not that difficult, you can use the existing I2C primitive methods to do it (probably, the following is untested):
```
import machine
class QuasiI2C(machine.I2C):
    def write(self, buf):
        try:
            super().write(buf)
        except OSError:
            # ignore errors from no ACK
            pass

    def writeto(self, addr, buf):
        tmp = bytearray(1)
        tmp[0] = addr << 1
        self.start()
        self.write(tmp)
        self.write(buf)
        self.stop()
```
Then just pass an instance of that I2C class to the SSD driver.

deshipu, 2016-08-29:
Did you get your display to work?

Personally I have used a number of different SSD1306-based displays, both SPI and I2C, and didn't have any problems with them. The ACK thing may be just one possible explanation, but you would get the same error basically with any communication problems, like wrong address, bad connection, lack of pullup resistors on the SPI line, etc.

Does `i2c.scan()` even show your display?

rlust, 2016-11-29:
Having the same error on an ESP8266.
Did you get this resolved?

deshipu, 2016-11-29:
Hi @rlust, does the device appear when you do `i2c.scan()`?

mkarliner, 2016-11-29:
In my case, excessively long wires on a breadboard made it fail in funny ways

deshipu, 2016-11-29:
Well, the longer the wires, the stronger (lower resistance) pullups you need. Until you reach the limit of what a pin can sink.

o0alexela0o, 2019-08-30:
> In my case, excessively long wires on a breadboard made it fail in funny ways

same for me. Thanks

jonnor, 2024-09-08:
No new information in over 5 years, and very likely to be a hardware/wiring issue.

closed 2024-09-08 by jonnor