micropython/micropython #2317: Esp8266: can't get the __neg__ operator to work

rguillon, 2016-08-13:
The following code works on the linux port but not on the esp8266: 

`class a():
    def __init__(self, val):
        self.val = val
    def __neg__(self):
        return a(-self.val)

a1 = a(5)
a2 = -a1
print(a2.val)
`

I got a `TypeError: unsupported type for : 'a'`

Did i missed something?

deshipu, 2016-08-13:
Yes. For performance reasons, MicroPython doesn't let you redefine operators.

rguillon, 2016-08-13:
I see, however __add__ and __sub__ seems to work fine.

pfalcon, 2016-08-13:
> The following code works on the linux port but not on the esp8266:

Should be fixed by 9cf29493569d6292fc672e757b2aa96b42f1fe1d

closed 2016-08-13 by pfalcon

rguillon, 2016-08-13:
It works, that was fast!

Thanks.

deshipu, 2016-08-13:
I'm sorry for speaking too soon, I really thought it's completely unsupported. Is there a list somewhere of which operators are supported?

pfalcon, 2016-08-13:
We support "all", except possibly for the esoteric ones, like matrix mul. There may be also some differences in semantics. A unit test for this functionality would a definitive reference of what's supported.