micropython/micropython #2323: unix: hashlib test failing depending on compiler used

deshipu, 2016-08-16:
I was asked to report this bug here, because the original reported can't/doesn't want to use github. He described his issue here: http://forum.micropython.org/viewtopic.php?f=3&t=2260

> hashlib/test_hashlib.py fails on Scaleway C1 ARM Linux system (comparable to Raspberry Pi 2). Running Debian 8, gcc 4.9.2
> 
> Micropython built from current master, commit info:
> 
> commit: 21967990519b3d0778770ddb0994216d012101c3
> Author: Damien George <damien.p.george@gmail.com>
> Date: Sun Aug 14 16:51:54 2016 +1000
> 
> I've debugged the python code a little. The error happens in the sha_final function, where in the last block (count= 19 chars) the function tries to zero out the last 45 chars of the block using sha_info['data'][:count] and is getting an empty array as this slice (sha_info['data'] is a list of 64 ints). So it looks like the slice is being made incorrectly.
> 
> The test works fine under micropython on x64 linux and under cpython 3.4.2 on the Scaleway ARM.

dpgeorge, 2016-08-16:
It looks like this is the same as #2264.  I fixed it in 095e43a9a5b3fc7c1220cdbabe6a4fed0973b1e1, so please try that.

deshipu, 2016-08-16:
That was indeed that problem, sorry for the duplicate.

closed 2016-08-16 by deshipu