micropython/micropython #2324: PWM issues when using OneWire library

thomasDOTde, 2016-08-16:
Using the OneWire library breaks all PWM outputs becaus they are interrupt driven.
This is due to massive delays inside of read_bit/write_bit functions disabling interrupts first.

    int32_t i = disable_irq();
    pin_set(pin, 0);
    DELAY_US(esp_onewire_timings[TIMING_READ1]);
    pin_set(pin, 1);
    DELAY_US(esp_onewire_timings[TIMING_READ2]);
    int value = pin_get(pin);
    enable_irq(i);

I'd suggest to use timer0 to do OneWire timing.

dpgeorge, 2016-08-17:
I assume you are talking about the esp8266 port.  That chip does not have many hardware features and so is limited in what it can do (eg onewire and PWM at the same time is difficult).  Do you have working, reliable code for a onewire driver that uses timer0?  Do you know of any other esp8266 projects that can do both PWM and timing critical things like onewire?

dpgeorge, 2017-06-14:
As mentioned above, doing both software PWM and software OneWire at the same time on the esp8266 is not easy.  The OneWire readbit function disables interrupts for about 10us, and writebit, for about 70us.  Using a Timer with IRQs would not be able to get the accuracy required for timing readbit/writebit.

If this is still an issue please investigate and suggest a way to improve it.

closed 2017-06-14 by dpgeorge