micropython/micropython #2325: Can't read sd card?

Chriswu92, 2016-08-17:
Hi,  I have a new pythonboard 1.0. After upgrade it.I reset it and find "pybflash" in DriverManager. Then I  insert a sd card into the slot, but nothing change. Is sth wrong with my slot? or maybe the .py? help.... (My OS is win7

dpgeorge, 2016-08-17:
You will need to restart the board for the SD card to become visible (it is not automatically "mounted" by the pyboard).  You can restart by pressing the RST button.

Chriswu92, 2016-08-17:
@dpgeorge  I restart the board again and again... but still pybflash , no sd card.

peterhinch, 2016-08-17:
With the SD card fitted, what happens if you type the following at the REPL:
```python
import os
os.listdir('/')
```

dpgeorge, 2016-08-17:
It could be that you have some dust in the SD connector, and/or dust on the contacts of the SD card.  Try cleaning it: wipe the card and put it in and out of the connector a few times.  To check that the SD card is present you can type: `pyb.SDCard().present()`.  If that returns `False` then the card is not being detected.

peterhinch, 2016-08-17:
The one time I had trouble with the SD card was after cleaning solder flux from the board with IPA. Some of the flux had got transferred to the connector. Further cleaning fixed it.

Chriswu92, 2016-08-17:
I clean both SD card and SD connector, then I use the  SD card in my mobile phone, it is working fine ,  maybe the SD connector is bad.

dpgeorge, 2016-08-17:
@Chriswu92 : what does `pyb.SDCard().present()` print when executed at the REPL with the SD card inserted?  If it prints False then the SD card switch is broken.

Chriswu92, 2016-08-18:
It prints False.

Chriswu92, 2016-08-18:
@dpgeorge : I found boot.py, main.py, pybcdc, and README.md 4 files in pybflash . need anything else?

dpgeorge, 2016-08-19:
Those 4 files are correct.  I saw a comment by you that `pyb.SDCard().present()` returns False, but that comment seems to be deleted.  What happened?  If that command returns False with an SD card inserted then the SD card switch is broken.  You can fix it by bending in the small metal tab on the bottom left corner of the connector (as viewed from the top of the pyboard).  Alternatively, email contact -at- micropython.org to arrange for a replacement board.

Chriswu92, 2016-08-20:
I replace another board which return True.But can't read SD card either. My friend said all boards I bought online had design defect.

dpgeorge, 2016-08-22:
How big is your SD card?  Try formatting it again.  Or try a different SD card.

All boards that are shipped have had a micro SD card plugged in to the socket and have been verified working.

dpgeorge, 2017-06-14:
@Chriswu92 did you solve this problem?

dpgeorge, 2017-06-26:
Assuming that this is no longer a problem.  If it is, reopen with more details.

closed 2017-06-26 by dpgeorge