micropython/micropython #2328: overload operator __rmul__ does not work

bolorkhuu, 2016-08-19:
i'm trying to test the overload operator `__rmul__` on micropython.

```
class foo:
    def __init__(self, data):
        self.data = data
    def __mul__(self, other):
        print("mul")
        if type(other) in (int, float):
            return foo(self.data * other)
        else:
            return foo(self.data * other.data)
    def __rmul__(self, other):
        print("rmul")
        if type(other) in (int, float):
            return foo(self.data * other)
        else:
            return foo(self.data * other.data)

if __name__ == '__main__':
    f1 = foo(10)
    f2 = foo(20)
```

on python

```

In [1]: (f1*f2).data
mul
Out[1]: 200

In [2]: (f1*50).data
mul
Out[2]: 500

In [3]: (50*f1).data
rmul
Out[3]: 500

```

Unfortunately it fails on micropython
```
>>> (f1*f2).data                                                                
mul                                                                             
200                                                                             
>>> (f1*50).data                                                                
mul                                                                             
500                                                                             
>>> (50*f1).data                                                                
Traceback (most recent call last):                                              
  File "<stdin>", line 1, in <module>                                           
TypeError: unsupported types for __mul__: 'int', 'foo'                         
```

Does anyone know how to solve it?

Thanks!

pfalcon, 2016-08-19:
Duplicate of http://forum.micropython.org/viewtopic.php?f=2&t=2287 . Please avoid double-posts.

closed 2016-08-19 by pfalcon

AaronKel, 2017-02-17:
Just incase anyone searches for this in the future as I did, the solution is as follows.
Replace runtime.c: generic_binary_op to
```
    // generic binary_op supplied by type
    mp_obj_type_t *typel;
	//#if MICROPY_PY_REVERSE_METHODS
	mp_obj_type_t *typer;
	//#endif
generic_binary_op:
    typel = mp_obj_get_type(lhs);
	//#if MICROPY_PY_REVERSE_METHODS
	typer = mp_obj_get_type(rhs);
	//#endif
    if (typel->binary_op != NULL) {
        mp_obj_t result = typel->binary_op(op, lhs, rhs);
        if (result != MP_OBJ_NULL) {
            return result;
        } 
		//#if MICROPY_PY_REVERSE_METHODS
		else {
			switch(op) {
                case MP_BINARY_OP_MULTIPLY: {
					mp_obj_t result2 = typer->binary_op(MP_BINARY_OP_REVERSE_MULTIPLY, rhs, lhs);
					if (result2 != MP_OBJ_NULL) {
						return result2;
					}
				}
				case MP_BINARY_OP_TRUE_DIVIDE: {
					mp_obj_t result2 = typer->binary_op(MP_BINARY_OP_REVERSE_TRUE_DIVIDE, rhs, lhs);
					if (result2 != MP_OBJ_NULL) {
						return result2;
					}
				}
				case MP_BINARY_OP_ADD: {
					mp_obj_t result2 = typer->binary_op(MP_BINARY_OP_REVERSE_ADD, rhs, lhs);
					if (result2 != MP_OBJ_NULL) {
						return result2;
					}
				}
				case MP_BINARY_OP_SUBTRACT: {
					mp_obj_t result2 = typer->binary_op(MP_BINARY_OP_REVERSE_SUBTRACT, rhs, lhs);
					if (result2 != MP_OBJ_NULL) {
						return result2;
					}
				}
				
				default:
                    goto unsupported_op;
			}
		}
		//#endif
    }
```
add to py/runtime0.c with in second typedef enum
```
	MP_BINARY_OP_REVERSE_MULTIPLY,
	MP_BINARY_OP_REVERSE_TRUE_DIVIDE,
	MP_BINARY_OP_REVERSE_ADD,
	MP_BINARY_OP_REVERSE_SUBTRACT,
```
add to py/objtype.c in const qstr mp_binary_op_method_name
```
	[MP_BINARY_OP_REVERSE_MULTIPLY] = MP_QSTR___rmul__,
	[MP_BINARY_OP_REVERSE_TRUE_DIVIDE] = MP_QSTR___rtruediv__,
	[MP_BINARY_OP_REVERSE_ADD] = MP_QSTR___radd__,
	[MP_BINARY_OP_REVERSE_SUBTRACT] = MP_QSTR___rsub__,
```
It's a very niche thing so I don't know if it is worth a pull request but it would be good to have it written down here (Incase I forget).
You can customise the \__r***__ methods to a build by uncommenting 
```
//#if MICROPY_PY_REVERSE_METHODS
```
and adding to py/mpconfig.c with
```
#ifndef MICROPY_PY_REVERSE_METHODS
#define MICROPY_PY_REVERSE_METHODS (0)
#endif
```

The \__r***__ functions will add an extra overhead when applying to any method that fails to provide a \__***__ method.

dpgeorge, 2017-04-07:
@AaronKel you should be able to make the code more compact by using a small look-up-table to convert the enum for the normal ops to reverse ops, then just having one call to typer->binary_op.

jimmo, 2019-10-30:
For anyone who finds this bug -- this was implemented in Sept 2017 in eb84a830df62813f5af7f0144fc77444bf18f3a8 but requires ` MICROPY_PY_REVERSE_SPECIAL_METHODS` to be enabled (currently only enabled by default in the Unix and Windows ports).