micropython/micropython #2329: Allow to disable python assertions

SquirrelCZE, 2016-08-19:
Is there possibility to somehow disable python assertions? I tried to google around this and there is bash parameter for micropython... but

how to achieve this effect for things like esp8266 ?

given that esp is target as not strong device, it would allow for easier development of code that pushes it's limits, because asserts can be great way to debug things, while slow down the esp...

dpgeorge, 2016-08-25:
No, there is no way at the moment to disable the `assert` statement (not even on unix port).

CPython's behaviour is that passing -O on the command line disables assert.  For us to emulate that behaviour would be easy, just don't compile assert statements if the mp_optimise_value parameter is non-zero.  That would be enough to disable asserts in the unix port (which already allows the -O option).

But then for bare metal ports, how to control the mp_optimise_value?  AFAIK there is no official way to do it in CPython when scripts are running.  The problem is that the optimisation only takes effect for the next script that is executed, not the current one.  I think we had this discussion before, and in the stmhal port you use `pyb.main("main.py", opt=1)` to enable optimisation.  We'd need to think of something like that but for all ports to use.

How about `micropython.optimise(value)` (or even just `opt`)?  The semantics would be that that function call just changes the mp_optimise_value variable to the given value.  You could then put this in boot.py, and then when main.py is executed it has optimisations enabled (ie assert disabled, and some other things).

dpgeorge, 2016-08-25:
Another option is `sys.optimise(value)` but we should talk to upstream before adding that.

SquirrelCZE, 2016-08-25:
Well, for me, adding this option into boot.py seems as good solution ...

as for exact semantics, I don't think I can really give a hand :)

dpgeorge, 2016-08-26:
With 24df30c1336fc72ad7dcd5cfc08a5e7f9a1f9808 assertions are now disabled when optimisations are turned on (eg on unix: ./micropython -O).

closed 2016-09-04 by pfalcon