micropython/micropython #2331: segfault when argument to btree.open() is not an object

blazewicz, 2016-08-20:
## Problem
Segmentation fault when argument to btree.open() is not an object.

### example code

```Python
>>> import btree
>>> btree.open('foo')
[1]    4779 segmentation fault  ./micropython
```

## Explanation

Problem lays in argument checking inside mod_btree_open (modbtree.c:354)

```C
    // Make sure we got a stream object
    mp_get_stream_raise(pos_args[0], MP_STREAM_OP_READ | MP_STREAM_OP_WRITE | MP_STREAM_OP_IOCTL);
```

which calls:

```C
const mp_stream_p_t *mp_get_stream_raise(mp_obj_t self_in, int flags) {
    mp_obj_base_t *o = (mp_obj_base_t*)MP_OBJ_TO_PTR(self_in);
    const mp_stream_p_t *stream_p = o->type->protocol;
    if (stream_p == NULL
        || ((flags & MP_STREAM_OP_READ) && stream_p->read == NULL)
        || ((flags & MP_STREAM_OP_WRITE) && stream_p->write == NULL)
        || ((flags & MP_STREAM_OP_IOCTL) && stream_p->ioctl == NULL)) {
        // CPython: io.UnsupportedOperation, OSError subclass
        nlr_raise(mp_obj_new_exception_msg(&mp_type_OSError, "stream operation not supported"));
    }
    return stream_p;
}
```

This function assumes that ```self_in``` is always an object and it only checks if it implements a _protocol_ and if this protocol supports requested operations.

Whenever first argument for btree.open() is not an object (but a qstr for example), a non pointer dereference happens inside mp_get_stream_raise and thus causes a segfault.

## Solution

The solution is to use ```MP_OBJ_IS_OBJ()``` macro either before ```mp_get_stream_raise()``` call in mod_btree_open or inside of the mp_get_stream_raise().

### fix example:

```Diff
diff --git a/py/stream.c b/py/stream.c
index 473eb96..81b995c 100644
--- a/py/stream.c
+++ b/py/stream.c
@@ -94,16 +94,24 @@ mp_uint_t mp_stream_rw(mp_obj_t stream, void *buf_, mp_uint_t size, int *errcode
 }
 
 const mp_stream_p_t *mp_get_stream_raise(mp_obj_t self_in, int flags) {
+    // check if self_in is an object before derefencing
+    if (!MP_OBJ_IS_OBJ(self_in)) {
+        goto err;
+    }
+
     mp_obj_base_t *o = (mp_obj_base_t*)MP_OBJ_TO_PTR(self_in);
     const mp_stream_p_t *stream_p = o->type->protocol;
     if (stream_p == NULL
         || ((flags & MP_STREAM_OP_READ) && stream_p->read == NULL)
         || ((flags & MP_STREAM_OP_WRITE) && stream_p->write == NULL)
         || ((flags & MP_STREAM_OP_IOCTL) && stream_p->ioctl == NULL)) {
-        // CPython: io.UnsupportedOperation, OSError subclass
-        nlr_raise(mp_obj_new_exception_msg(&mp_type_OSError, "stream operation not supported"));
+        goto err;
     }
     return stream_p;
+
+    err:
+        // CPython: io.UnsupportedOperation, OSError subclass
+        nlr_raise(mp_obj_new_exception_msg(&mp_type_OSError, "stream operation not supported"));
 }

-- 
2.7.4
```

after fix:

```Python
>>> import btree
>>> btree.open('foo')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: stream operation not supported
```

If everyone likes this solution I can prepare a PR.

pfalcon, 2016-08-20:
Nice write-up, I like the headings ;-).

> either before mp_get_stream_raise() call in mod_btree_open or inside of the mp_get_stream_raise().

So, now feel free to wear maintainer's hat and decide which of these two are better, preferably with arguments.

blazewicz, 2016-08-20:
thanks :)

I've made ```mp_get_stream_raise``` similar to ```mp_get_buffer[_raise]```.

pfalcon, 2016-08-20:
Thanks too, but is that the only argument? Doing something like someone/something else isn't always a good argument on its own. Good arguments would come from considering "if I make a one, or another change, what outcomes of it will be?". In either case, the commit message isn't good - it's unclear how switching one for another avoids segfault. The description why it helps, and "made mp_get_stream_raise similar to mp_get_buffer[_raise]." should really be in the commit message.

blazewicz, 2016-08-20:
Ok, you're right. Is it good now?

pfalcon, 2016-08-23:
Generally, next expected step was to analyze all call sites of mp_get_stream_raise() and make sure it's not on hot paths (like, in loops). Well, this is small useful change, so I'm merging it as is. Thanks for contribution!

commit 65620764 (2016-08-23): py/stream.c: use mp_obj_get_type in mp_get_stream_raise

In current state `mp_get_stream_raise` assumes that `self_in` is an object
and always performs a pointer derefence which may cause a segfault.

This function shall throw an exception whenever `self_in` does not implement
a stream protocol, that includes qstr's and numbers.

fixes #2331
closed 2016-08-23 by pfalcon