micropython/micropython #2335: esp8266: Support up to 16 MB of flash

Ratfink, 2016-08-21:
The ESP8266 supports up to 16 megabytes of SPI flash [(source, page 12)](https://espressif.com/sites/default/files/documentation/0a-esp8266ex_datasheet_en.pdf).  NodeMCU has supported this much flash for about two months now, and I notice that MicroPython does not.  I don't know all the nuances of the ESP8266, but from what I understand there are three things that should all encode the same size: the flash chip's ID, the SpiFlashChip structure from the SDK, and the bootloader size ID.  They might not all agree about the flash chip's size, so we try to make them all agree in flashbdev.py.  It looks like in the similar routine from NodeMCU, they use size IDs 8 and 9 for 8 MB and 16 MB, respectively.  Setting these IDs in MicroPython seems to make mpy think that it has only 512 kB of flash, and I'm not sure why.  Ignoring the bootloader size ID and only calculating flash size from the flash chip's ID seems to work, but somehow that seems like the wrong thing to do.

Long story short, I want to add support for 8 and 16 MB of flash to MicroPython, but I don't know what the right way to do it is.  Any advice would be greatly appreciated.

dpgeorge, 2016-08-22:
You'll want to extend the pattern here to account for 8 and 16 mb: https://github.com/micropython/micropython/blob/master/esp8266/modules/flashbdev.py#L30-L39 .  And then change the 22 to 24 on this line: https://github.com/micropython/micropython/blob/master/esp8266/modules/flashbdev.py#L53

Ratfink, 2016-08-22:
I tried making the suggested change:

```diff
diff --git a/esp8266/modules/flashbdev.py b/esp8266/modules/flashbdev.py
index 98b0208..284fa1a 100644
--- a/esp8266/modules/flashbdev.py
+++ b/esp8266/modules/flashbdev.py
@@ -37,6 +37,10 @@ def set_bl_flash_size(real_size):
         code = 3
     elif real_size == 4096*1024:
         code = 4
+    elif real_size == 8192*1024:
+        code = 5
+    elif real_size == 16384*1024:
+        code = 6
     else:
         code = 2
     buf = bytearray(4096)
@@ -50,7 +54,7 @@ def set_bl_flash_size(real_size):
 size = esp.flash_id() >> 16
 # Check that it looks like realistic power of 2 for flash sizes
 # commonly used with esp8266
-if 22 >= size >= 18:
+if 24 >= size >= 18:
     size = 1 << size
     if size != esp.flash_size():
         import machine
```
It results in a boot loop, printing this repeatedly:

    l��r��  yr�'b����x�p�lp�������p�lll��r��        yr�'b��l����x��p�������p��cl`䈌������rn������x���b���b����ߌ��b��bl`�>r�8�l`#��r�����8����r�����8�����lrr��9l�Ğ����br�b�������b�#�n��1�n��`ll������l��l�����;�n��Β��b���b�>��ܒ��b�|lrrr�n�����|~����       6��������b����xbl`��l�#4 ets_task(40100390, 3, 3fff6300, 4)
    Bootloader Flash size appear to have been set incorrectly, trying to fix
    bcn 0
    del if1
    usl
    
     ets Jan  8 2013,rst cause:2, boot mode:(3,6)
   
    load 0x40100000, len 30860, room 16 
    tail 12
    chksum 0x17
    ho 0 tail 12 room 4
    load 0x3ffe8000, len 1060, room 12 
    tail 8
    chksum 0xd7
    load 0x3ffe8430, len 3000, room 0 
    tail 8
    chksum 0xb1
    csum 0xb1

If I set it to use code 8 for 8 MB and 9 for 16 MB, like NodeMCU does, I get a similar boot loop, but with more stuff being printed each time it boots:

```
�����������r������r�l�������������8������
���l����l���l����l��������l�����������܌���rl��r��       yr�'b�����x�p�lp�������p�lll��r��       yr�'b��l
����8��p�������p��cl`��������rn������8���b���b�����ߌ��b��bl`�>r�x��l`#��r��#3 ets_task(40100390, 3, 3fff
6300, 4)
Fatal exception 9(LoadStoreAlignmentCause):
epc1=0x40250a35, epc2=0x00000000, epc3=0x00000000, excvaddr=0x0000001b, depc=0x00000000

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 30860, room 16
tail 12
chksum 0x17
ho 0 tail 12 room 4
load 0x3ffe8000, len 1060, room 12
tail 8
chksum 0xd7
load 0x3ffe8430, len 3000, room 0
[... 135 lines cut ...]
        self.writebyte(MATCH_ROM)
        self.write(rom)

    def scan(self):
        devices = []
        diff = 65
        rom = False
        for i in range(0xff):
            rom, diff = self._search_rom(rom, diff)
            if rom:
                devices += [rom]
            if diff == 0:
                break
        return devices

    def _search_rom(self, l_rom, diff):
        if not self.reset():
            return None, 0
        self.writebyte(SEARCH_ROM)
        if not l_rom:
            l_rom = bytearray(8)
        rom = bytearray(8)
        next_diff = 0
        i = 64
        for byte in range(8):```

dpgeorge, 2016-08-25:
Try erasing the flash entirely before deploying the image.  It looks like the ESP SDK is finding the wifi/system parameters in the wrong location.

Ratfink, 2016-08-25:
I erased the flash before deploying the image every time above.  It does seem like it's reading the system parameters from the wrong location.  From [this pull request](https://github.com/nodemcu/nodemcu-firmware/pull/1390) from NodeMCU, it looks like the ESP SDK doesn't really support more than 4 MB of flash even though the chip itself does.  Maybe it would be better to not worry about supporting this much flash until Espressif themselves do.

dpgeorge, 2016-08-26:
> it looks like the ESP SDK doesn't really support more than 4 MB of flash even though the chip itself does. Maybe it would be better to not worry about supporting this much flash until Espressif themselves do.

Yes.  We can revisit this issue when upgrading to the latest ESP SDK, which has a new way of locating and storing the persistent wifi/config parameters.

mcauser, 2016-08-31:
FYI - New WeMos D1 Mini Pro has 16MB flash
http://www.aliexpress.com/store/product/WEMOS-D1-mini-Pro-16M-bytes-external-antenna-connector-ESP8266-WIFI-Internet-of-Things-development-board/1331105_32724692514.html

peterhinch, 2016-08-31:
Is that kosher? It doesn't appear on http://www.wemos.cc/

mcauser, 2016-08-31:
Yeah it's legit. Was only released a few hrs ago. It'll be on their website soon, I guess.

deshipu, 2016-10-19:
So now that we use the new sdk, I tried this again, but I'm still getting a boot loop.

deshipu, 2017-02-24:
There is some hope, the next version of SDK should support it: https://github.com/espressif/esptool/issues/123#issuecomment-281256779

RockyTV, 2017-10-28:
Any update on this?

soocki, 2017-11-09:
Yes any updates regarding this? It's been a while now. No hope I guess.

BoKKeR, 2017-12-10:
Any update? The SDK seems to support it

whatnick, 2018-01-25:
I installed v2.1.0 of esp-open-sdk . Compiled upy and still cannot load firmware with 16MB flash set, works with 4MB just fine.

robert-hh, 2018-01-25:
What is returned by esp.flash_size()?

whatnick, 2018-01-25:
I am stuck in boot loop after flashing with 16M as flash size, works fine
with 4M.

dpgeorge, 2018-02-02:
Thanks to @whatnick I now have a 16M board and was able to work on this issue.  I managed to get uPy working with the full 16M but it required 3 things:
1. Update ESP SDK to v2.1.0 or later.
2. Update esptool.py to a recent version that supports 16M flash.
3. After erasing the flash (make erase), manually flash the esp_init_data_default.bin file (found in the ESP SDK release) to the location 0xffc000.  This can be done using the command: `esptool.py --port /dev/ttyUSB0 write_flash --verify 0xffc000 esp_init_data_default.bin`

Then it seems to work and `esp.flash_size()` returns 16M.  Also the filesystem uses up to the end of the flash.

For flash sizes up to 4Mbytes, uPy can automatically write the esp_init_data_default.bin data to the correct location in the flash.  But for 16M flash chips this automatic step seems to fail because of 2 reasons: 1) the flash size is incorrectly reported as 4M; 2) SPIRead and SPIWrite can't seem to access beyond 4Mbytes.  My guess is that the full range of 16Mbytes of flash requires special handling and is "unlocked" during the Espressif boot code, which happens too late for the uPy automatic writing part.

So it might be that the only way to support 16M of flash is to require the user to manually flash the esp_init_data_default.bin file (of course we can provide a convenient "make" target to do this automatically).  I'm not sure how Arduino/NodeMCU deal with this.

whatnick, 2018-02-02:
I have been flashing the 16M boards with Arduino provided esptool.exe
binary. May be they forked esptool and ship a custom version.

whatnick, 2018-02-09:
> esptool.py --port /dev/ttyUSB0 --baud 115200 write_flash
> *--flash_size=4MB* 0 esp8266-20171101-v1.9.3.bin
>
> Work fine .

But the point is you lose the rest of the 16MB flash. Thanks for the proper
fix.

keijsvog, 2018-02-11:
Hi @dpgeorge,

Can you describe the procedure step-by-step for my D1 Mini Pro? I first followed your steps:

1. updates SDK to 2.1.0
2. updated ESPtool
3. Erase flash by `esptool.py --port /dev/cu.SLAB_USBtoUART erase_flash `
4. Executed: `esptool.py --port /dev/cu.SLAB_USBtoUART write_flash --verify 0xffc000 esp_init_data_default.bin`

Next I flashed Micropython:
`esptool.py --port /dev/cu.SLAB_USBtoUART --baud 460800 write_flash -fm dio -fs 16MB 0 esp8266-20171101-v1.9.3.bin
`
Unfortunately this does not work because the board keeps resetting. What am I doing wrong?

Another question: when I download the newest SDK 2.2.0, there is a esp_init_data_default_v05.bin and a esp_init_data_default_v08.bin, which on to use to flash instep 4?

Thanks in advance.

dpgeorge, 2018-02-13:
> Unfortunately this does not work because the board keeps resetting. What am I doing wrong?

@keijsvog did you rebuild MicroPython and use that firmware (rather than one from https://micropython.org/download)?  The point is that you need to build the ESP8266 firmware using the new SDK; see `ports/esp8266/README.md` for more details about building it.

> there is a esp_init_data_default_v05.bin and a esp_init_data_default_v08.bin, which on to use to flash instep 4?

I'm not sure.  Try one and if it doesn't work then try the other :)

keijsvog, 2018-02-14:
@dpgeorge,

Thanks it worked! Short description for those how struggle with the same challenge:

First of all you should have a ESP8266 board with 16MB memory. I have the Wemos D1 Pro. The description below work on a Mac.

1. Start with upgrading the SEPTools: `pip install esptool --upgrade`

2. Compile the esp-open-sdk by following the [instructions](https://github.com/pfalcon/esp-open-sdk/) This will download during make the 2.1.0 SDK (at least at time of writing these instructions) You will need the file esp_init_data_default.bin later on (the file is located in the directory esp-open-sdk/ESP8266_NONOS_SDK-2.1.0-18-g61248df/bin/)  For your convenience, copy the file to a separate directory.

3. Compile the MicroPython port to ESP8266 by following the [instructions](https://github.com/micropython/micropython/blob/master/ports/esp8266/README.md) 

4. You will end up with a file called firmware-combined.bin which you can also copy to the directory already containing esp_init_data_default.bin. Change to this directory

5. First erase the ESP8266: `esptool.py --port /dev/cu.SLAB_USBtoUART --baud 460800 erase_flash`

6.  Flash both the micropython firmware and the esp_init_data_default.bin: `esptool.py --port /dev/cu.SLAB_USBtoUART --baud 460800 write_flash -fm dio -fs 16MB 0 firmware-combined.bin 0xffc000 esp_init_data_default.bin`

7. Open a serial terminal to your ESP8288 (e.g. with the serial monitor included in the Arduino IDE) and reset the ESP8288

8. Type into the serial terminal following commands:
```
>>> import esp
>>> esp.flash_size()
```

I get following result:
`16777216`

So now we can use the full 16MB of the board!

soocki, 2018-02-14:
#3613

soocki, 2018-02-15:
Do you think it is worth to create a paragraph in 'ports/esp8266/README.md' pointing to this thread and specifically the answer above regarding the chips with more than 4MB flash? There's a lot of them out there, a paragraph like that would be quite helpful I would imagine.

mcauser, 2018-02-22:
I pip upgraded my esptool to the latest version, however, I was still getting errors trying to connect at baud 460800 and using -fs 16MB. Turns out, the older esptool 1.x in my esp-open-sdk volume was being called.

In my `~./bash_profile`:
`export PATH=/Volumes/case-sensitive/esp-open-sdk/xtensa-lx106-elf/bin:$PATH`

```
$ which esptool.py
/Volumes/case-sensitive/esp-open-sdk/xtensa-lx106-elf/bin/esptool.py
```

Commented out the `export PATH`, opened a new terminal. Ran esptool write_flash, success!!

```
>>> esp.flash_size()
1677721
```
🍻

alanjds, 2018-02-27:
@keijsvog Can you please share your `firmware-combined.bin`? Then the ones that want not to go thru the process just to see if it works can try it.

keijsvog, 2018-02-27:
@alanjds,

Attached is a zip file with the two files needed: `firmware-combined.bin` and `esp_init_data_default.bin`

I hope this works for you.
[MicroPython-Firmware-esp8266.zip](https://github.com/micropython/micropython/files/1764650/MicroPython-Firmware-esp8266.zip)

alanjds, 2018-02-27:
Thanks a lot.

I had not worked. I will try to use some old binaries reported to be working, even if on 4MB. I had not yet made to work on Mini Pro on any version.

alanjds, 2018-03-01:
Looks like the 4MB version worked. I got baud-rate issues to test some days ago.

Will try your binary again later..

flashypepo, 2018-03-01:
@keijsvog: thank you for the two binairies! It works on a WeMOS D1 mini Pro with 16MB after following your instructions. The esptools.py I used is version 2.3.1. Will try to build the binary firmware-combined.bin later.

alanjds, 2018-03-01:
Now it worked! I had some mismatch on the baud rate of the serial terminal connection :S 

Worked:
```bash
$ picocom /dev/tty.SLAB_USBtoUART --baud 115200
```

devxpy, 2018-09-11:
Here are the upto date firmaware images.

[wemos d1 mini pro.zip](https://github.com/micropython/micropython/files/2371187/wemos.d1.mini.pro.zip)

matthiasheng, 2018-11-16:
Thanks all for the hard work! I got 16MB works fine here!

gretel, 2019-05-09:
thanks folks 😻just flashed a `wemos d1 mini pro` to support 16MB of flash.

```shell
$ esptool.py version
esptool.py v2.6
2.6
$ esptool.py --port ${PORT} --after no_reset --baud 460800 erase_flash
# ./esp-open-sdk/ESP8266_NONOS_SDK-2.1.0-18-g61248df/bin/esp_init_data_default.bin
$ esptool.py --port ${PORT} write_flash 0xffc000 esp_init_data_default.bin
# do not autodetect, force 16MB flash
$ esptool.py --port ${PORT} --baud 460800 write_flash -fm dio -fs 16MB 0x00000 firmware-combined.bin
```

```python
import esp
esp.flash_size()
16777216
```

MarSoft, 2019-07-24:
Right now it works if I just flash the latest firmware version from downloads **and** `esp_init_data_default.bin` from https://github.com/espressif/esp8266_at/tree/master/bin - first erase flash, then flash them both, just like @gretel suggested in the comment above.

flashypepo, 2019-07-24:
Also confirmed.  It works just like @gretel suggested. Thanks.
Used esptool.py version 2.5.1 - Lolin D1 mini Pro v2.0.0 - esp.flash_size() -> 16MB!

xlee2000, 2019-07-30:
Thanks very much.

reillybeacom, 2019-08-11:
I would also like to **confirm** this new strategy works and I have been waiting for something like this *"proper"* solution for a long time! Thanks @gretel !

It is still slightly confusing that the answer is spread over the last couple of comments so I would like to combine them here:

Download `esp_init_data_default.bin` from here (click "View raw"):

https://github.com/espressif/ESP8266_AT/blob/master/bin/esp_init_data_default.bin

Download the latest version of MicroPython for 8266 from here:

http://micropython.org/download#esp8266

Erase the flash:

     esptool.py --port ${PORT} --after no_reset --baud 460800 erase_flash

Flash esp_init_data_default.bin:

    esptool.py --port ${PORT} write_flash 0xffc000 esp_init_data_default.bin

Flash to the 16mb ESP8266 module (eg. Lolin D1 Mini Pro) for your version of `esp8266-20190529-v1.11.bin`:

    esptool.py --port ${PORT} --baud 460800 write_flash -fm dio -fs 16MB 0x00000 esp8266-20190529-v1.11.bin

Can anyone explain what `esp_init_data_default.bin` is and does? Any downsides?

Can anyone suggest a good place in the documentation for this? (So as to maybe close this issue with a PR.) eg. https://github.com/micropython/micropython/blob/master/docs/esp8266/tutorial/intro.rst

Thanks

MJRuffDevelop, 2019-10-08:
Been looking at this myself and I cannot for the life of me, get it to work.

robert-hh, 2019-10-08:
Did you flash both the firmware image and esp_init_data_default.bin?
Can you flash the device in 4MB mode, by specifying the flash size, option: -fs 4MB
If you, you may have have to try other flash modes.

mycognosist, 2019-12-10:
A slight variation of the approach suggested by @gretel : 

I received an error on the `write_flash` command for `esp_init_data_default.bin`, since `esptool.py` was auto-detecting the flash size as 4MB. I had to specify the `-fs 16MB` flag in the first flash command to achieve success:

`esptool.py --port /dev/ttyUSB0 write_flash 0xffc000 esp_init_data_default.bin -fs 16MB`

Thanks to all.

luhua2018, 2021-01-13:
[image]

fe328, 2021-02-11:
So the trick stopped working in micropython 1.14. Anyone have any ideas? =)

gretel, 2021-02-12:
@fe328 what version of `esptool` are you using?

erongd, 2021-02-23:
> @fe328 what version of `esptool` are you using?

I tried both esptool v3.0 and esptool v2.8, neither works with v1.14. With v1.13 it works just fine for Wemos D1 mini Pro v2 with 16MB flash, exactly the same command.

gretel, 2021-03-06:
@erongd the [changelog of micropython 1.14](https://github.com/micropython/micropython/releases/tag/v1.14) reads:

```
The esp8266 port now includes three generic board definitions: GENERIC_512K, GENERIC_1M and GENERIC, with the latter for modules with 2M or more of flash. This port also changed the default filesystem to littlefs, and no longer supports hard pin IRQs. See breaking changes below for more details.
```

think you have to build using a board defintion of `GENERIC` now.

erongd, 2021-03-17:
> the [changelog of micropython 1.14](https://github.com/micropython/micropython/releases/tag/v1.14) reads:
>
> ```
> The esp8266 port now includes three generic board definitions: GENERIC_512K, GENERIC_1M [...]

Yes rebuilding the firmware with "GENERIC" seems to have worked. However I thought the default build for 2M+ board is 
"GENERIC" what is different?

gretel, 2021-03-17:
> Yes rebuilding the firmware with "GENERIC" seems to have worked. However I thought the default build for 2M+ board is
> "GENERIC" what is different?

the supported flash size as stated in the changelog.

MatthiasLohr, 2023-01-01:
Again, 2.5 years later, no progress?

robert-hh, 2023-01-01:
There have been various instruction on how to enable 16M support, like https://github.com/micropython/micropython/issues/2335#issuecomment-520210822. I re-checked using this procedure to upload a fresh Micropython GENERIC build after erasing flash. It works.