micropython/micropython #2338: esp8266: can't unmount filesystem -- segfault

deshipu, 2016-08-22:
Since I found a way to mount an sd card in place of the internal filesystem, I was looking for a way to unmaount a mounted filesystem, so that people don't need to edit their ``_boot.py`` to use an sd card.

I found code in https://github.com/micropython/micropython/blob/master/extmod/fsusermount.c#L53-L70 that suggests that to unmount a file system, I just need to pass ``None`` as the device. However, when I try that on the esp8266, I get a reboot:

```
>>> uos.VfsFat(None, "")
Fatal exception 29(StoreProhibitedCause):
epc1=0x4025aa26, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 30860, room 16 
tail 12
chksum 0xff
ho 0 tail 12 room 4
load 0x3ffe8000, len 1060, room 12 
tail 8
chksum 0x33
load 0x3ffe8430, len 3000, room 0 
tail 8
chksum 0x55
csum 0x55
```

dpgeorge, 2016-08-22:
It won't work this way because you are trying to construct a new VfsFat instance, with a null device.  It probably does unmount, but then "segfaults" because the constructor is given a null vfs object after the unmount operation (which it then assumes is an actual object and dereferences it).

There doesn't seem to be a way (even undocumented) to unmount a device on the esp8266 port.  Proper handling of filesystems depends on #2140.

deshipu, 2016-08-22:
I tried to make a separate method that would unmount it:

```
STATIC mp_obj_t fat_vfs_unmount(mp_obj_t vfs_in) {
    mp_obj_t args[2] = {
        mp_const_none,
        mp_const_empty_bytes,
    };
    fatfs_mount_mkfs(2, args, (mp_map_t*)&mp_const_empty_map, false);
    return mp_const_none;
}
STATIC MP_DEFINE_CONST_FUN_OBJ_1(fat_vfs_unmount_obj, fat_vfs_unmount);
```

but it fails with the same error.

deshipu, 2016-08-22:
No, sorry, it fails with a slightly different traceback:
```
Fatal exception 28(LoadProhibitedCause):
epc1=0x40107686, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 30860, room 16 
tail 12
chksum 0x9c
ho 0 tail 12 room 4
load 0x3ffe8000, len 1060, room 12 
tail 8
chksum 0x23
load 0x3ffe8430, len 3000, room 0 
tail 8
chksum 0x45
csum 0x45
```

deshipu, 2016-08-22:
The code for this is at https://github.com/deshipu/micropython/commit/327dbbec84a4a4927e779e9bbe9f1011362f8a49

dpgeorge, 2016-08-22:
Try using the fsuser_umount_obj function in extmod/fsusermount.c.

deshipu, 2016-08-22:
I get `AttributeError: 'VfsFat' object has no attribute 'unmount'`.

deshipu, 2016-08-22:
Ok, that actually didn't crash. I got:
```
>>> uos.unmount()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 22] EINVAL
```

and the filesystem is still mounted, though.

deshipu, 2016-08-22:
Looks like it's not finding the device in the list?

deshipu, 2016-08-22:
Passing a device instead of path made it work. However, the code I used to actually do it all is super-ugly, I wonder if there is a better way to call that function, than removing the STATIC from it?

https://github.com/deshipu/micropython/commit/5afb16d71ea66f4d3fdb4d23ffae3c7c35395c3d

deshipu, 2016-08-22:
I found a nice way to expose the umount function, the pull request is at https://github.com/micropython/micropython/pull/2344

closed 2016-08-26 by dpgeorge

pfalcon, 2016-08-26:
I didn't have a chance to review this, and this is not accompanied to changes to https://github.com/micropython/micropython/blob/master/tests/extmod/vfs_fat_ramdisk.py , so further changes may be required (including non-compatible API changes to ensure compatibility among the ports).

deshipu, 2016-08-26:
What changes do you have in mind?

pfalcon, 2016-08-26:
The idea of OO VFS interface that it works on any port, including unix, where there's a native OS mounting. So, there should be clear separation in OO VFS API (which should work everywhere, and in the same manner), and "uos" module emulation on top of it (which is needed only for some ports). This patch is visible improvement in that regard comparing to the previous patch, thanks. But I'm still not sure this presents well-defined interface for 2-stage unmounting described above. I can't say what changes may be required, just until we have OO VFS API finalized, they may be. Testing on unix port is a way to ensure that simpler of the omissions are not made.

deshipu, 2016-08-26:
Great. I can try and help to implement the changes when you actually know what you want. In the mean time, we have a working `uos.umount()` that lets us use an SD card on the ESP8266.

Do you think the `uos.umount()` function will need to be removed or change semantics, or do the "non-compatible API changes" refer to something else, that I fail to see?

pfalcon, 2016-08-26:
What I'm 100% sure now is that VfsObj.umount() call should be exercised in https://github.com/micropython/micropython/blob/master/tests/extmod/vfs_fat_ramdisk.py , so if you have time, you're welcome to look into that. (For future cases, I'd actually think there should be a requirement that VFS changes accompanied by a working example (i.e. test) of running on unix port). Note that VFS support is enabled in coverage build for unix (not normal build).

dpgeorge, 2016-08-26:
The semantics of mounting anyway need to change.  For example, uos.VfsFat(bdev, '/ramdisk') should not need the second argument because a mount point doesn't make sense in the OO scheme.

deshipu, 2016-08-26:
> What I'm 100% sure now is that VfsObj.umount() call should be exercised in https://github.com/micropython/micropython/blob/master/tests/extmod/vfs_fat_ramdisk.py , so if you have time, you're welcome to look into that.

I added a simple test in https://github.com/micropython/micropython/pull/2360, however it fails at the moment, because we've only added the uos.umount for the ESP8266, and not the Unix port.

pfalcon, 2016-08-26:
And that's the point I mention above - there should not be uos.umount() in unix port (if it existed, it would call OS' system call). It should be just vfs.umount() (where vfs is VFS object created above). The rest of the test is exactly right - unmount, test it worked, mount again and see that it worked and gives expected results.

deshipu, 2016-08-26:
By the way, I noticed, that using a vfs that has been unmounted results in a crash. I wonder if we should have a flag and a check in all the methods, or avoid it in some other way.

pfalcon, 2016-08-26:
Well, the way you coded #2360 seems to work, and further improvements in that area would go with #2140.

commit d29ca282 (2016-08-26): esp8266/modous: Add os.umount method to unmount a filesystem.

This is an object-oriented approach, where uos is only a proxy for the
methods on the vfs object.  Some internals had to be exposed (the STATIC
keyword removed) for this to work.

Fixes #2338.