micropython/micropython #2339: ESP8266 non-fatal error bcn_timeout

peterhinch, 2016-08-22:
I've seen this once (two occurrences in succession) in many tens of hours of total runtime:
```
bcn_timout,ap_probe_send_start
bcn_timout,ap_probe_send_start
```
The application hung for a period but then continued running normally. This leads me to suspect that the hang occurred in a socket operation but I have no means of proving that. The period of hanging was on the order of tens of seconds, but again it's hard to be precise because I didn't spot the exact time when the problem began.

The application is performing repeated publications using mqtt.simple at qos == 0.

I appreciate this isn't much to go on, but alas it's all I have.

pfalcon, 2016-08-23:
So, what outcome is expected to this report?

peterhinch, 2016-08-24:
I wondered if it was a symptom of an issue. Is policy not to raise as issues vendor debug notices?

pfalcon, 2016-08-24:
No, it's diagnostic information. Or otherwise Espessif, Inc. is the only party who would know for sure. Did you ask them?

peterhinch, 2016-08-25:
No. This the first of a few cryptic messages I've seen in many hours of testing. They don't affect my application. I thought they might offer useful information to you guys if they were perhaps provoked by the MicroPython runtime. If such reports serve no purpose I'm happy to comply.

pfalcon, 2016-08-25:
Thanks. Well, we've seen them all (if not in reality, then in disassembly), so it doesn't offer much useful info on its own, unless accompanied by steps how to reproduce it or what it affects (and then it's still more of academic interest generally, as again, that's Espressif stuff).

peterhinch, 2016-08-25:
Fair enough, point taken.

closed 2016-08-25 by peterhinch