micropython/micropython #2340: ESP8266 hangs on MQTT Connection

Trebastian, 2016-08-22:
As descibed in this thread:
http://forum.micropython.org/viewtopic.php?f=16&t=2269
the system hangs after some time in my wlan network. If error logging is on I get a Fatal exception 28. Sometimes the chip restarts after that. The restart was on a system running the esp8266-20160809-v1.8.3.bin version producing this output:

Fatal exception 28(LoadProhibitedCause):
epc1=0x401021d6, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000

ets Jan 8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 30872, room 16
tail 8
chksum 0x5d
load 0x3ffe8000, len 1064, room 0
tail 8
chksum 0x0b
load 0x3ffe8430, len 3000, room 0
tail 8
chksum 0x76
csum 0x76
d▒▒|▒▒{oc▒▒
l▒
c▒l
#䄜▒▒▒c▒lsll
l▒▒|▒▒rrgc▒▒l
▒▒▒▒
c▒
c쌜▒▒▒b▒▒drl▒d▒▒|▒▒{oc▒▒l▒▒l
c▒
c쌜▒▒▒
c▒lcd▒▒s▒l▒l▒d`▒▒s▒d▒d▒d`▒▒{▒$
▒▒▒
ll`rl▒▒sl▒▒▒cd▒c{|▒cxc
▒▒o▒gg▒
ld▒▒l쎌
l▒▒ll▒▒▒d▒▒l▒n▒▒▒▒cd▒pğ▒c
l
▒cpslsd{▒g▒▒▒
▒csc▒▒c쎜▒▒cdcddb▒b▒▒n▒▒▒▒c{c▒▒c䏜▒▒cl
l
lcl▒
l▒▒|▒▒rrgc▒▒
l▒▒▒c▒
b䄜▒▒▒▒▒bddcl▒▒do▒l`▒▒lo▒psl
▒l▒▒|▒▒#5 ets_task(40100390, 3, 3fff6300, 4)
WebREPL daemon started on ws://192.168.4.1:8266
WebREPL daemon started on ws://0.0.0.0:8266
Started webrepl in normal mode

But most of the time the system just hangs without any reaction or restart.

Trebastian, 2016-08-28:
Just an update. I flashed 3 Wemos modules with the actual version:
MicroPython v1.8.3-65-g263aaa7 on 2016-08-28
I saw that the watchdog is not enabled and now the boards usually reset and reboot after they hang. 
The error messages look like this in the REPL:
Board 1:

Fatal exception 28(LoadProhibitedCause):
epc1=0x40102232, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 30964, room 16
tail 4
chksum 0x41
load 0x3ffe8000, len 1064, room 4
tail 4
chksum 0x05
load 0x3ffe8430, len 3000, room 4
tail 4
chksum 0x12
csum 0x12
l▒▒|▒▒s{oc▒▒
l▒c▒l
c서▒▒▒
c▒dsll
l▒▒|▒▒s{oc▒▒l
▒▒x▒c▒▒
c쌜▒▒▒c▒
▒l{d▒d▒▒|▒▒ss'b▒▒d▒▒▒▒c▒c䌜▒▒▒b▒dcl▒▒r▒l▒d▒
▒l`▒▒s▒d▒$▒
l`▒▒{▒l
▒▒▒
dd`{l▒▒{l▒▒▒b
l▒b{|▒cx▒▒ccc▒▒o▒gg▒dl▒▒l쎌d▒▒$l▒▒▒l▒▒l▒o▒▒▒▒cd▒p
▒▒▒cd▒cp{d{dr▒n▒▒▒
▒cx▒▒▒▒▒
▒▒c▒d▒▒c
▒c▒▒n▒▒▒▒cpb▒▒▒▒▒▒▒c
▒l▒▒bd`▒
l▒▒|▒▒s{oc▒▒
l▒▒▒b▒
c▒ܜ▒▒▒▒cl
lcd▒▒do▒l`▒▒do▒xsl
▒l▒▒|▒▒{oc▒▒
d▒#5 ets_task(40100390, 3, 3fff6300, 4)
WebREPL daemon started on ws://192.168.4.1:8266

Board 2:

Fatal exception 28Fatal exception 28Fatal exception 28
 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x40100000, len 30964, room 16
tail 4
chksum 0x41
load 0x3ffe8000, len 1064, room 4
tail 4
chksum 0x05
load 0x3ffe8430, len 3000, room 4
tail 4
chksum 0x12
csum 0x12
l▒▒|▒▒rrgc▒▒
l▒b▒dc䄜▒▒▒
c
▒l{ddd▒▒|▒▒{oc▒▒l
▒▒p▒
c▒
c쌜▒▒▒
c▒
▒l{d▒
l▒▒|▒▒ssgb▒▒d▒▒▒▒b▒b䄜▒▒▒c▒dcl▒▒{▒l▒l▒
▒l`▒▒r▒l▒l▒
l`▒▒s▒l
▒▒▒
ll rl▒▒sl▒▒▒cd▒c{|▒cxc
▒▒o▒o▒
ll▒▒l쇌
l▒▒ll▒▒▒l▒▒d▒g▒▒▒▒cd▒p▒▒▒cd▒cxslsls▒o▒▒▒
▒
rc▒▒c쇜▒▒blbl
lc▒c▒▒g▒▒▒▒cs
▒▒c쎜▒▒cdcddbl▒d▒▒|▒▒{oc▒▒$▒▒▒c▒
c쌜▒▒▒▒▒
cl
lcd▒▒lg▒l`▒▒lg▒p;l▒d▒▒|▒▒ssnb▒#5 ets_task(40100390, 3, 3fff6300, 4)
WebREPL daemon started on ws://192.168.4.1:8266

Board 3:

Fatal exception 28(LoadProhibitedCause):
epc1=0x40102232, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000

 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x40100000, len 30964, room 16
tail 4
chksum 0x41
load 0x3ffe8000, len 1064, room 4
tail 4
chksum 0x05
load 0x3ffe8430, len 3000, room 4
tail 4
chksum 0x12
csum 0x12
l▒▒|▒▒{oc▒▒d▒
▒db䄜▒▒▒b▒l{dl
l▒▒|▒▒{oc▒▒d▒▒p▒
c▒
c쌜▒▒
c▒▒dsl▒d▒▒|▒▒ssgc▒▒l▒▒▒▒c▒
c서▒▒▒b▒dbd▒▒s▒l▒l▒
▒l`▒▒{▒d▒d▒d`▒▒s▒l
▒▒▒
ll`sl▒▒{l▒▒▒cd▒b{|▒bp▒▒bbc▒▒o▒o▒dl▒▒d䏄l▒▒ll▒▒▒l▒▒l▒o▒▒▒▒cd▒p▒▒▒c
l
▒cp{d{ls▒o▒▒▒▒
{c▒▒c䎜▒▒cl▒
l▒l▒c▒▒o▒▒▒▒c{c▒▒c䏜▒▒cl▒
l▒l{d▒
l▒▒|▒▒{oc▒▒d▒▒▒c▒
c쌜▒▒▒▒▒
cl
dcl▒▒dn▒d`▒▒lg▒p{d
▒l▒▒|▒▒{oc▒▒#5 ets_task(40100390, 3, 3fff6300, 4)
WebREPL daemon started on ws://192.168.4.1:8266

pfalcon, 2016-10-20:
Sorry for the lack of progress with this, this is due to lack of exact information how to reproduce it. We recently upgraded to ESP8266 SDK 2.0, which is known to fix at least some of older issue. Please try to reproduce this issue with the the latest master (requires latest esp-open-sdk), and see if the issue still persists. If it does, please attach firmware.bin, firmware.elf, firmware.map files for investigation. Thanks.

Trebastian, 2016-10-27:
I can reproduce the failure with the newest firmware on two different boards running the same attached firmware version. Both Boards are connected to the same network. But i don't think the problem is related to MQTT. I first noticed the failure while running MQTT samples. But the board hangs or reboots also appear if I'm just connected to the wlan and doing nothing. Perhaps a special network package or something.

First reboot happend while running a MQTT sample program:

>>> Fatal exception 28
 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x40100000, len 29972, room 16
tail 4
chksum 0xc7
load 0x3ffe8000, len 1084, room 4
tail 8
chksum 0x89
load 0x3ffe8440, len 3000, room 0
tail 8
chksum 0x66
csum 0x66
l▒▒|▒▒{oc▒▒d▒
▒db䄜▒▒▒l
c
▒l{dl
l▒▒|▒▒rrgc▒▒l
▒▒▒c▒b䌜▒▒▒
c▒
▒lsl▒d▒▒|▒▒ssnc▒▒l▒▒▒
▒c▒
c서▒▒b▒lcd▒▒s▒l▒l▒
l`▒▒{▒d▒d
l`▒▒r▒l
▒▒▒
dd`rl▒▒sl▒▒▒c
l▒cs|
▒cxc▒▒n▒gg▒
ld▒▒l쇌d▒▒dd▒▒▒l▒▒d▒g▒▒▒▒c
l▒p▒▒▒cl
▒cp{d{d{▒g▒▒▒
▒
{c▒▒c䎜▒
▒ll▒▒l▒l▒c▒▒o▒▒▒▒b{c▒▒c쎜▒
▒ll▒▒l▒lsl▒d▒▒|▒▒ssgc▒▒
l▒▒
▒c▒
c䄜▒▒▒▒▒bl
lcd▒▒lo▒l`▒▒do▒xrl▒l▒▒|▒▒rrgc▒▒
▒#5 ets_task(40100164, 3, 3fff6310, 4)
WebREPL daemon started on ws://192.168.4.1:8266
WebREPL daemon started on ws://0.0.0.0:8266
Started webrepl in normal mode

Second one happend while running nothing. Only connected via com port:

>>> Fatal exception 28
 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x40100000, len 29972, room 16
tail 4
chksum 0xc7
load 0x3ffe8000, len 1084, room 4
tail 8
chksum 0x89
load 0x3ffe8440, len 3000, room 0
tail 8
chksum 0x66
csum 0x66
l▒▒|▒▒{oc▒▒d▒
▒db䄜▒▒▒l
c
▒l{dl
l▒▒|▒▒rrgc▒▒l
▒▒▒c▒b䌜▒▒▒
c▒
▒lsl▒d▒▒|▒▒ssnc▒▒l▒▒▒
▒c▒
c서▒▒b▒lcd▒▒s▒l▒l▒
l`▒▒{▒d▒d
l`▒▒r▒l
▒▒▒
dd`rl▒▒sl▒▒▒c
l▒cs|
▒cxc▒▒n▒gg▒
ld▒▒l쇌d▒▒dd▒▒▒l▒▒d▒g▒▒▒▒c
l▒p▒▒▒cl
▒cp{d{d{▒g▒▒▒
▒
{c▒▒c䎜▒
▒ll▒▒l▒l▒c▒▒o▒▒▒▒b{c▒▒c쎜▒
▒ll▒▒l▒lsl▒d▒▒|▒▒ssgc▒▒
l▒▒
▒c▒
c䄜▒▒▒▒▒bl
lcd▒▒lo▒l`▒▒do▒xrl▒l▒▒|▒▒rrgc▒▒
▒#5 ets_task(40100164, 3, 3fff6310, 4)
WebREPL daemon started on ws://192.168.4.1:8266
WebREPL daemon started on ws://0.0.0.0:8266
Started webrepl in normal mode

[Firmware.zip](https://github.com/micropython/micropython/files/555144/Firmware.zip)

pfalcon, 2016-10-28:
> with the newest firmware

For reference, that's esp8266-20161025-v1.8.5-49-g760ed46 (based on the contents of the attached archive).

dpgeorge, 2022-06-25:
Closing due to inactivity.

closed 2022-06-25 by dpgeorge