micropython/micropython #2342: ESP2866: Maximum Deep Sleep is 35 minutes

dries007, 2016-08-22:
I've noticed that the deep sleep mode (or I assume any RTC alarm) only handles up to 35 minutes (2 100 000 milliseconds) as input value. After that it seems to overflow.

After some lookup work I found [this](http://blog.falafel.com/esp8266nodemcu-deep-sleep/) article, which mentions that the maximum for SDK 1.5.1 is 71 minutes (4 260 000 milliseconds).

Given those values and the information in the article, I believe that the value passed to the `RTC.alarm` function is multiplied internally to get the required microsecond based value. If that multiplication is done with unsigned variables instead of signed ones, the same 71 minutes maximum would be achievable.

I'm willing to hunt for this myself if required, but I'm unfamiliar with this codebase and I don't have the toolchain yet.

If, in the meantime, there is a way I can bypass this cap (writing to a mem location), please let me know. Thanks!

Sidenote, could I, to make longer delays possible, use a raw memory location inside the RTC memory to make a loop counter? I don't want to use the flash for this if I can avoid it.

dpgeorge, 2016-08-25:
We have support for detecting RTC overflow so that we use a full 64-bit number to represent time (in microseconds).  The alarm also uses 64-bits.  So in principal it should allow the maximum sleep time that the hardware supports, ie 71 minutes.

I fixed a bug in the 64-bit arithmetic in d09b6b9aa1c2ed82a4795f5abbc02f50f00927d9 which should get it working properly, but I didn't check it actually sleeping for 71 mins.  Can you check that it works now?

dries007, 2016-08-25:
The overflow detection seems to be broken now, and from 2 tests it seems like it can do 71 minutes now. I set it to 90 minutes and by the time my alarm went off (set it for 71 minutes) the console was back up.
I'm doing more tests now, but they all take a while :smile:

dries007, 2016-08-25:
Another, better timed test:
I set the deep sleep for 70 minutes, and after 62 minutes it already rebooted. Is that due to inaccurate RTC or is it a software issue?

dpgeorge, 2016-08-26:
> I set the deep sleep for 70 minutes, and after 62 minutes it already rebooted. Is that due to inaccurate RTC or is it a software issue?

Probably due to RTC inaccuracies.  The software itself will try to do the largest deepsleep it can when possible, see https://github.com/micropython/micropython/blob/master/esp8266/modmachine.c#L109-L115

dries007, 2016-08-26:
I saw that code before, but when I tried using more then 35 minutes of sleep with the old  firmware it acted like the time had already expired and instantly restarted.
Now it seems to always do the longest time correctly. Though I think an exception would be better in that scenario. Besides that I think this issue is resolved.

pfalcon, 2016-09-20:
Closing then.

closed 2016-09-20 by pfalcon